Bodmer/TFT_eSPI #963: DMA on ILI9488

inb4ohnoes, 2021-01-29:
Hi!

From what I can gather, DMA is not supported on ILI9488 with the esp32 running this library. Is this a technical issue? Or is it just something you haven't supported yet? Or maybe I'm missing something? For my application I'd really love to get that working...

Bodmer, 2021-01-29:
Yes, that is correct. The reason is mostly historical.

The library uses 16 bits per pixel colour (565 format, 5 bits red, 6bits green and 5 bits blue) as this is compatible with the majority of the low cost displays available, so 2 bytes per pixel are sent to the display.

Displays typically support a range of bits per pixel options. In parallel mode the ILI9488 supports 16 bits per pixel, but sadly in SPI mode it requires 18 bits per pixel. The  18bits must be sent as 3 bytes, with the most significant 6 bits of each byte representing the red, green and blue. This is rather inconvenient for two reasons, firstly the transfer of pixels to the display is 50% slower (3 bytes instead of 3 per pixel) and also the 16 bits colours must be converted to 3 bytes using masking and shifts, which also slows the transfer to display further.  The display update is also slowed because the library is not so well optimised for pixel transfer of images to 18 bits per pixel displays. Recently I tested  a more optimised approach and this is ~3x faster so this will be in the next library update coming soon. This may give you the performance boost you need without resorting to DMA.

To DMA to an ILI9488 a new pixel buffer needs to be creatred 1.5x the size of the original buffer, then all 16 bit pixels coverted to 18 bits in 3 bytes, then the existing DMA functions can be used. This could be done in a sketch rather than the library.

Bodmer, 2021-01-29:
P.S. Eventually Sprites will support 24 (888 RGB format) and 32 bits per pixel (ARGB format where A is the alpha (transparency)). This is moving the library towards the higher performance larger RAM capacity micro controllers. This evolution of the library will happen gradually as more performant low cost microcontrollers become available at very low cost.

inb4ohnoes, 2021-01-30:
Thank you for the detailed explanation. This is exactly what I was looking for. In this case, I'll hold off on trying to get DMA working on 9488 and wait for your more optimized approach. 3x faster is more than enough for my application. How soon do you think this update will be? I'm dying to see how fast it will be 😅

Bodmer, 2021-01-30:
I have uploaded a[ Beta test branch](https://github.com/Bodmer/TFT_eSPI/tree/2.4.0-Beta).

This can only be downloaded via that Github link, not from the Arduino IDE library manager.

Make sure you keep a copy of any setup files before you upgrade so thay do not get over-written.

The performance boost is for Sprite and image rendering improvements not functions lie fillRect. So jpegs are now rendered much faster and the SpriteRotatingCube example with DMA turned off now runs at 70fps instead of 20fps.

Let me know how you get on with it. Further testing is needed but I have not found any problems yet.

I have also added a new anti-aliased sketch, plus drawWideLine and drawWedgeLine anti-aliased  functions which are mailnly for rotating hands/needles on dials.

closed 2021-02-01 by Bodmer

inb4ohnoes, 2021-02-01:
Thank you so much for your continued work and the test branch. I will test it and let you know how it goes. I really appreciate your promptness and responsiveness. It's something that is really under-appreciated these days...

inb4ohnoes, 2021-02-02:
I just tried the sprite demos on this beta, and wow the performance increase is awesome! It's hard to believe you can get this much performance out of this hardware. This is exactly what I needed, thank you so much!

Bodmer, 2021-02-02:
I am pleased to here this!
Thanks for your kind comments, they are appreciated.

funkypoly, 2021-04-19:
Please excuse me if i am a bit out of topic . I am trying to find a definitive answer regarding the ILI948X screens.
I bought one ILI9481 3.5'' because i wanted an IPS screen. My question is simple. Are all ILI948X screens IPS ? Is ILI9486 & 9488 IPS ???

PS. I tried the beta library with my ILI9481 and it's looking nice and fast ! Thanks !

andrey-belokon, 2022-09-27:
> I have uploaded a[ Beta test branch](https://github.com/Bodmer/TFT_eSPI/tree/2.4.0-Beta).

This branch is not available - error 404. Readme note that DMA not supported for ILI9486/9488 SPI. Is this still correct?

PS I ask because now need make choice between 9341 and 9486/88. Wide display is more preferable for my project, but DMA required.