Bodmer/TFT_eSPI #964: 3bit/idle mode with st7789 SPI

mahmeh, 2021-01-29:
Hey, i cant find if the idle/3bit mode is supported, but would be great if it was
I really like to use to use a color tft as monochrome, or atleast be very sparse with the colors, and have good frame rate by sending as little data as possible.

Thx

Bodmer, 2021-01-29:
I am not sure what support is being requested. Here is a guess.

The display can be commanded into the idle mode using library commands, e.g.:
```
tft.writecommand(ST7789_IDMON);  // Idle mode on
tft.writecommand(ST7789_IDMOFF); // Idle mode off
```

The display does not have a 3 bit colour mode at the SPI interface level. This is an internal function invoked by the idle commands. Please refer to the ST7789 datasheet for more information.

closed 2021-01-29 by Bodmer

mahmeh, 2021-01-30:
Thanks, i wanted to use the idle mode as i only need 8colors in my project and felt that i could get a performance increase by just sending 3bits of data/pixel over spi. maybe there is another way of doing this. I was going for a display with the ILI9488 controller which seems to support this, but the display i wanted was not available with the ILI9488