Bodmer/TFT_eSPI #965: setTextSize not working with smooth fonts

MrFox42, 2021-01-29:
tft.setTextSize() doesn't seem to be working when I'm using smooth fonts.

Here's the simplified loop of the Font_Demo_1 sketch for a quick test:

```
void loop() {
  tft.fillScreen(TFT_BLACK);
  tft.setTextColor(TFT_WHITE, TFT_BLACK); // Set the font colour AND the background colour
                                          // so the anti-aliasing works
  tft.setCursor(0, 0); // Set cursor at top left of screen
  tft.loadFont(AA_FONT_SMALL); // Must load the font first
  tft.println("Small 15pt font"); // println moves cursor down for a new line
  tft.setTextSize(5);
  tft.println("Small 15pt font"); // println moves cursor down for a new line  
  delay(5000);
  tft.unloadFont(); // Remove the font to recover memory used
}
```

Both lines printed with the same font size. Is this intentional?

Bodmer, 2021-01-29:
I understand that this might be an expected capability but setTextSize will not work with smooth fonts and is not intended to. I will add a note to the examples. To use a larger Smooth Font font it has to be created as a vlw file and added to the sketch.

This is because there is no simple way to enlarge the anti-aliased bitmap and achieve the same quality of the anti-aliasing effect.

MrFox42, 2021-01-30:
Thank you for the quick response. I understand, just wanted to make sure it's intentional and not a bug / user error.

The main reason I wanted to use _setTextSize_ is because the file size of the font converted to .vlw got pretty large above ~100 pixels. But I realized I only need a specific set of characters on this pixel size for my project, so I can convert the font with just those few characters included.

Bodmer, 2021-11-21:
ReadMe updated.

closed 2021-11-21 by Bodmer