Bodmer/TFT_eSPI #968: ILI9481 troubles ... not working

Koxx3, 2021-01-30:
Hello !

I am struggling with my ILI9841.
Vendor link : https://fr.aliexpress.com/item/32829704426.html?spm=a2g0s.9042311.0.0.7e9d6c372FZSoE

I am using an FPC adapter.

I wired the screen with this configuration :
[image]

WARNING : I am not sure about the wiring ... chinese documentation... Is SPI_A0 = TFT_DC ?

I launched the "Read_User_Setup" with this result : 
`
[code]
TFT_eSPI ver = 2.3.59
Processor    = ESP32
Frequency    = 240MHz
Transactions = Yes
Interface    = SPI
Display driver = 9481
Display width  = 320
Display height = 480

MOSI    = GPIO 23
MISO    = GPIO 19
SCK     = GPIO 18
TFT_CS   = GPIO 22
TFT_DC   = GPIO 2
TFT_RST  = GPIO 17

Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled

Display SPI frequency = 20.00
[/code]`

Nothing is visible on the screen with this one or another sketch.

Any help please ?

Koxx3, 2021-01-30:
2 more informations :
- I tested IM1 at 3.3V and GND, no change
- I use "Setup21_ILI9488" but I forced ILI9481

Koxx3, 2021-01-30:
ok, it finally works. It was a speed issue. I cannot go higher thant 15MHz. may be the wire lenght (20cm).

Bodmer, 2021-01-30:
Well done for finding the solution so fast.

The display [data sheet here](https://www.displayfuture.com/Display/datasheet/controller/ILI9481.pdf) page 138 indicates a maximum rated speed of 10MHz (100ns write cycle time) so it looks like the ILI9481 cannot be overclocked by much.  Some more recently designed displays (ST7796)  have a higher rated speed ~25MHz but work reliably at 40MHz (and sometimes up to 80MHz).

Long wires can have an effect and a short (or multiple) GND lines help. IF thet are long (>150mm) try reducing wires to 100mm or less if you can. It may not make much difference though if the limiting factor is the ILI9481 display.

Bodmer, 2021-01-30:
You will be able to update the display faster if you use 8 bit parallel connections since the write cycle is still 100ns (10MHz), then 8 bits are being loaded in the same period instead of 1. The library does not support 16 bit parallel interfaces.

ILI9481 displays are available in UNO format and can be plugged into a UNO style ESP32. If you do this, note the modifications given in the ReadMe.

Koxx3, 2021-01-30:
thanks a lot for your answer..i already shorten the wires to 10cm
centimeters. no change.

i also found the 100ns in the datasheet.

i was surprised because i could overclock the ili9341 to 60 mhz. ahahha.

once again, that's a lot for your answers. i trashed the adafruit lib to
use your lib 😉

interesting the ST7796 !

i am looking for an ips 3.5 inch screen with fast spi refresh (at least
40mhz, and no need for 'high bits colors'). any idea if i can find somthing
like that on aliexpress ?

Koxx3, 2021-01-30:
oh yes, and for the 8 bits parallel interface, impossible, I already use
all the IOs of the ESP32 😂

Bodmer, 2021-01-30:
It looks liek display prices have increased significantly over the las tfew months due parts shortages. This is the ST7796S SPI display I mentioned.  Unfortunately these seem scarce these days and the ones that are available have doubled in price. There is a design fault on them but this is quite easily fixed,[ see discussion here](https://github.com/Bodmer/TFT_eSPI/discussions/898).

The [RPi MHS 4.0 display](http://www.lcdwiki.com/MHS-4.0inch_Display-B) is good too as it has a controller than seems to work at 80MHz SPI rate with the ESP32 (40MHz with ESP8266).

Koxx3, 2021-01-30:
ok, if it's only the touch driver, I don't care, I don't need it ;)
I'll look at the RPi MHS.

thanks for your feedback.

closed 2021-01-30 by Bodmer

beckmx, 2022-10-11:
@Koxx3 hello there, the pinout for the IM0, IM1 etc were left as in the original post?, I noticed I bought a screen and I was given the same table u did.