Bodmer/TFT_eSPI #971: [Question] 16 bits color mode on ILI9481 / 88

Koxx3, 2021-01-31:
Hello !

In the ILI9481 datasheet, I see that the drivers support 16bits color mode :
[image]

and it's forced to 18 bits in the lib.
[image]

Would it be possible to change to 16 bits mode ?

Of course, I tried, to force SPI_16BIT_DRIVER and the display is ugly ;)

Koxx3, 2021-01-31:
hummmm ... DBI Type C is limited to 18bits ? 

[image]

Bodmer, 2021-01-31:
Yes, with a SPI interface the display does not support 16 bit colours. In parallel mode it does.

closed 2021-01-31 by Bodmer