Bodmer/TFT_eSPI #973: "TFT.deinit()" missing or unknown to me

kiromaisus, 2021-02-01:
Hi.
I'm using ESP32 TTGO T-Display and I'm sure by experiments with battery and amperimeter that the function init() is doing something that are not reversed by other functions (that I know).
It prevents from consuming low current in deep sleep (330 uA), instead consuming around 7,2 mA.
Is there a fuction to deinit the display?

Bodmer, 2021-02-01:
The display can be commanded into the idle mode using library commands, e.g.:
```
tft.writecommand(ST7789_IDMON);  // Idle mode on
tft.writecommand(ST7789_IDMOFF); // Idle mode off
```
This may reduce the power consumption. There are other commands that might reduce the power, see the ST7789 data sheet for more options.

The big current user tends to be the LED backlight, the TTGO T-Display has a backplight pin so turn it off when not needed.

kiromaisus, 2021-02-02:
Thank you for the answer.
Although it didn't work. I did comment the line "spi.begin(...)" inside tft.init() and the consumption ceased. So it is the spi that are still enabled and consuming. Also I saw that you didn't implement a function to really end the spi conversation. Sadly I am not capable of understanding how to do that.

kiromaisus, 2021-02-02:
I went around with a lot of stuff to discover that it was only needed this:

```
    digitalWrite(TFT_RST, HIGH);
    digitalWrite(TFT_RST, LOW);
    delay(1);
    digitalWrite(TFT_RST, HIGH);
```
A inverted reset to the reset pin. That's it.

Bodmer, 2021-02-03:
OK, according to the data sheet IDLE mode is a lower power mode but it does not say by how much.

Another option is to try the sleep mode (which is entered by a reset anyway):
```
tft.writecommand(TFT_SLPIN);    // Sleep in
tft.writecommand(TFT_SLPOUT); // Sleep out
```