Bodmer/TFT_eSPI #978: ILI9488 on esp32 doubt

beckmx, 2021-02-02:
Hello, I have a working PCB for the **ILI9341**, so I decided to replace it with a **ILI9488**, since both displays have the same pin location I thought that only by updating the setup.h file would be enough, but my screen seems it doesnt respond properly, looks like it shows like a gray shodow and fades out slowly, this is my setup file:

```

#define ILI9488_DRIVER

#define TFT_MISO 19
#define TFT_MOSI 23
#define TFT_SCLK 18

#define TFT_CS   26  // Chip select control pin
#define TFT_DC    5  // Data Command control pin

#define TFT_RST   -1  // Reset pin (could connect to RST pin)

#define TOUCH_CS 17     // Chip select pin (T_CS) of touch screen esp32 4m ram

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

#define SMOOTH_FONT

#define SPI_FREQUENCY  27000000
#define SPI_READ_FREQUENCY  16000000

#define SPI_TOUCH_FREQUENCY  2500000
```

Now,  I saw some comments about not connecting the same **SDO**, in my case both the display and the touch film are sharing this pin, do I need to unconnect one of them?

Bodmer, 2021-02-03:
Make sure [these lines](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setup_Select.h#L118-L123) are in your User_Setup_Select file.

Test with a defined reset pin, the ESP32 RST pin may not be sufficient to reset the display, for example:

#define TFT_RST   2

Or another free pin, and connect that pin to the TFT reset pin.

Check the compiler is loading the right setup file by looking at the serial output from the Read_User_Setup diagnostic sketch, post output here is you are not sure.

Bodmer, 2021-02-03:
Lastly yes, disconnect the SDO pin from the TFT. Use the Test_Touch_Controller diagnostic sketch to check the touch screen is working.

beckmx, 2021-02-03:
Ok, I disconnected the SDO, and tested the touch, and was working, but the display still not turning on (I tested the color test), this is the output you asked me for:

[code]
TFT_eSPI ver = 2.3.59
Processor    = ESP32
Frequency    = 240MHz
Transactions = Yes
Interface    = SPI
Display driver = 9341
Display width  = 240
Display height = 320

MOSI    = GPIO 23
MISO    = GPIO 19
SCK     = GPIO 18
TFT_CS   = GPIO 26
TFT_DC   = GPIO 5
TOUCH_CS = GPIO 17

Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled

Display SPI frequency = 27.00
Touch SPI frequency   = 2.50
[/code]

I also tested unwiring the reset pin, but still no color

Bodmer, 2021-02-03:
The display driver reported in the diagnostic output is 9341 ie ILI9341 it should say 9488
Check the setup file you are calling up has:

#define ILI9488_DRIVER

beckmx, 2021-02-03:
yup, surprising, so, I had to remove:

`// #include <User_Setup.h>           // Default setup is root library folder`

Looks like the compiler was reading the first flag of that file

closed 2021-02-03 by beckmx