Bodmer/TFT_eSPI #979: ESP32 + ILI9486 parallel + Touch compile error

Invertedlight, 2021-02-02:
I'm getting this compile error when adding #define TOUCH_CS to user_setup file.
Then compiling Touch_Controller_demo.

In file included from ../Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp:4595:0:
../Arduino/libraries/TFT_eSPI/Extensions/Touch.cpp: In member function 'void TFT_eSPI::begin_touch_read_write()':
../Arduino/libraries/TFT_eSPI/Extensions/Touch.cpp:22:34: error: 'spi' was not declared in this scope
     if (locked) {locked = false; spi.beginTransaction(SPISettings(SPI_TOUCH_FREQUENCY, MSBFIRST, SPI_MODE0));}

Without #define TOUCH_CS I get this error:
/var/folders/5r/wjc7xz2s72s70y559dkr9p6h0000gn/T/arduino_modified_sketch_987398/Touch_Controller_Demo.ino: In function 'void setup()':
Touch_Controller_Demo:44:9: error: 'class TFT_eSPI' has no member named 'setTouch'
     tft.setTouch(calibrationData);
         ^
Touch_Controller_Demo:47:9: error: 'class TFT_eSPI' has no member named 'calibrateTouch'
     tft.calibrateTouch(calibrationData, TFT_WHITE, TFT_RED, 15);
         ^
/var/folders/5r/wjc7xz2s72s70y559dkr9p6h0000gn/T/arduino_modified_sketch_987398/Touch_Controller_Demo.ino: In function 'void loop()':
Touch_Controller_Demo:64:11: error: 'class TFT_eSPI' has no member named 'getTouch'
   if (tft.getTouch(&x, &y)) {
           ^
exit status 1
'class TFT_eSPI' has no member named 'setTouch'

Any idea as to what I am doing wrong in the setup?
Graphics Demos work well and blazing fast.

#define ESP32_PARALLEL
#define ILI9486_DRIVER
// The ESP32 and TFT the pins used for testing are:
#define TFT_CS   33  // Chip select control pin (library pulls permanently low
#define TFT_DC   15  // Data Command control pin - must use a pin in the range 0-31
#define TFT_RST  32  // Reset pin, toggles on startup

#define TFT_WR    4  // Write strobe control pin - must use a pin in the range 0-31
#define TFT_RD    2  // Read strobe control pin

#define TFT_D0   12  // Must use pins in the range 0-31 for the data bus
#define TFT_D1   13  // so a single register write sets/clears all bits.
#define TFT_D2   26  // Pins can be randomly assigned, this does not affect
#define TFT_D3   25  // TFT screen update performance.
#define TFT_D4   17
#define TFT_D5   16
#define TFT_D6   27
#define TFT_D7   14

#define SD_SCK 	18
#define SD_MISO 19
#define SD_MOSI 23
#define SD_CS 	5

//#define TOUCH_CS 33

Bodmer, 2021-02-03:
The library does not support touch in parallel mode but alternative touch libraries ([see discussion](https://github.com/Bodmer/TFT_eSPI/discussions/895)) can be used

These have no meaning to the TFT_eSPI library and do not belong in the setup file unless they are for your sketch use (better to place in sketch in that case):

#define SD_SCK 18
#define SD_MISO 19
#define SD_MOSI 23
#define SD_CS 5

Invertedlight, 2021-02-03:
Hi Bodmer,

Thank you for the quick response.
I looked at the reference discussion you listed. Not sure this will work using the defines in the example as those PIN numbers are not in the parallel setup which uses all of the board pins.

#define CS_PIN  22
// 19 (MISO->T_DO), 23 (MOSI->T_DIN), 18 (SCK->T_CLK)

Defining CS_PIN 33 to match LCD_33 possibly? But then what about T_DO, T_DIN, T_CLK?

ESP32 Pin Mapping

LCD_RST 32

LCD_CS 33

LCD_RS 15    // == LCD_DC

LCD_WR 04

LCD_RD 02

LCD_D0 12

LCD_D1 13

LCD_D2 26

LCD_D3 25

LCD_D4 17

LCD_D5 16

LCD_D6 27

LCD_D7 14

[Image.jpeg]

Plugging the display into a Uno or Mega I can get touch working with the examples supplied by the vendor.  http://www.lcdwiki.com/3.5inch_Arduino_Display-UNO

I haven’t been able to figure out their library settings / map to pins on the ESP32.
Their sketches fail during compile using ESP32, but work fine with UNO & MEGA boards.
So it should be possible to read Touch using the listed pins in the parallel setup given an ESP32 compatible library.

Still scratching my head on this one.

James

Invertedlight, 2021-02-03:
[X]

Retrying image insert.

Bodmer, 2021-02-08:
If your display board is an UNO format and it has a resistive tough interface, then it can be used with an UNO format ESP32 as described in the library ReadMe and with the change made in this library. The resistive touch interface requires analaogue interface pins to be used and these pins are shared with the TFT display. It is rather complicated but from the pictures you should be able to work out the required connections.

The simplest solution is to buy an UNO format ESP32 board and make the wiring changes suggested. With an UNO format display it is then a plug-and-play setup.

Invertedlight, 2021-02-08:
Thanks!  I was coming to the same conclusion. The modifications seem simple enough.

I did try mapping to the tft control pins with the library. Running the example touch sketch I saw output with X maxed and the other values zero.  Remember seeing something about that and MISO/MOSI and not connecting one of them.

I’ll experiment more remapping to analog pins if possible on the ESP32 while I wait for a UNO ESP32 to arrive.  Also ordering a SPI display to play around with.  If I can get the ESP32 parallel to work I will post my findings.

Appreciate the help.

James

Bodmer, 2021-02-08:
These UNO displays do not have an SPI touch chip, so TOUCH_CS should not be defined.

The touch library you need and the modification [details are here.](https://github.com/s60sc/Adafruit_TouchScreen).

Bodmer, 2021-02-14:
Assumed resolved.

closed 2021-02-14 by Bodmer

Invertedlight, 2021-02-14:
Hi, no progress yet on my side. Been out sick with strep throat after getting the first COVID-19 vaccination. Finally feeling up to working on this some more. I’ll update later in the week.

Invertedlight, 2021-02-14:
Tried the other libraries suggested with no luck on the touch activation.
Moving on to the ESP32UNO board and the parallel TFT mods suggested.
Seems this will get things working for display and touch.

Also ordering a SPI TFT to try out with the ESP32-WROOM board.
Thanks for the help!