Bodmer/TFT_eSPI #980: Taking screenshots while using SD & touch ?

sn00zerman, 2021-02-04:
Hello Bodmer,

I want to use the screenshot functionality, to take some screenshots for my user-manual of my new little device.
(and to embed in the youtube movie serie that I'm creating)

However, how would one do this, while using the SD-card & touch ?
It is stated that I can't connect the MISO pin from the TFT, while using the SD & Touch, however, this is needed to readback data from the screen.
Also, in the User_Setup.h, I only have the "TFT_MISO" definition (that is used for the SD & Touch)

What extra (hidden) parameter can I use to connect the MISO of the TFT, while using the SD & Touch ?
Or is there another "trick" that will do ?

Thank you,
Kris

Bodmer, 2021-02-05:
Which processor and screen are you using.

sn00zerman, 2021-02-05:
Hi Bodmer,

I use ESP32 (LiliyGo TTGO T8 - Ebay object# 183692606067) 
and ILIILI9488 (320x480 via SPI) with touch and SD-card built-in.(Ebay object# 224157973027)
If serial communication does not work, it's no problem, I have enought space left on my SD-card to write the images to.

This is a part of my user_setup.h
#define TFT_MISO 19
#define TFT_MOSI 23
#define TFT_SCLK 18
#define TFT_CS   15 
#define TFT_DC    2  
#define TFT_RST   4
#define TOUCH_CS 13
#define TFT_BL   32         
#define TFT_BACKLIGHT_ON HIGH

Thanks in advance,
Kris

Bodmer, 2021-02-08:
It is possible to make the MOSI pin bidirectional at the ESP32 and ILI9488 so that the display can be read using readPixel(x, y) for example.  The changes necessary are, [change this line](https://github.com/Bodmer/TFT_eSPI/blob/master/TFT_Drivers/ILI9488_Init.h#L68) to:
```
#ifdef TFT_SDA_READ
    writedata(0x80);
#else
    writedata(0x00);
#endif
```
Add this line:
```
#define TFT_SDA_READ
```
To your user setup file, add the line after defining the display driver.

Change the SPI read frequency to 16MHz maximum in your user setup file:
```
#define SPI_READ_FREQUENCY  16000000 // May need to reduce to 10MHz if pixel colours are occasionally not corect.
```
You can now use the display read functions.

sn00zerman, 2021-02-09:
Thank you 👍
I will only have time on saturday to test this out.
I will keep you informed 🤗

sn00zerman, 2021-02-12:
I tested it today, but it does not work :-(
- I changed the code, like you explained above, until this point, everything works.
- When I connect the MISO from the TFT, to the MISO like defined in my user setup file, my SD-card and Touch do no longer work :-(
Any suggestions ?

Bodmer, 2021-02-12:
You must not connect the TFT MISO pin. The purpose of the changes above is to make the MOSI pin bidirectional so the single pin can be used to both write to and read from the display.

The edits look fine. I tried the changes and they work OK for me so I am not sure what the problem is.

Run the Read_User_Setup diagnostic example to make sure the compiler is picking up your changed settings. This is my output:
```
TFT_eSPI ver = 2.4.0
Processor    = ESP32
Frequency    = 240MHz
Transactions = Yes
Interface    = SPI
Display driver = 9488
Display width  = 320
Display height = 480

MOSI    = GPIO 23
MISO    = GPIO 19
SCK     = GPIO 18
TFT_CS   = GPIO 15
TFT_DC   = GPIO 2
TFT_RST  = GPIO 4
TOUCH_CS = GPIO 21

TFT_BL           = GPIO 32
TFT_BACKLIGHT_ON = HIGH

Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled

Display SPI frequency = 40.00
Touch SPI frequency   = 2.50
```

Bodmer, 2021-02-12:
Here is a simple TFT read/write pixel test sketch:
```
#include <TFT_eSPI.h> // Hardware-specific library
#include <SPI.h>

#define TDELAY 0

TFT_eSPI tft = TFT_eSPI(); 

void setup() {
  Serial.begin(115200);

  tft.init();
  tft.fillScreen(TFT_BLACK);
}

void loop() {
  static uint32_t wr = 1;
  static uint32_t rd = 0xFFFFFFFF;

  delay(TDELAY);

  tft.drawPixel(30,30,wr);
  Serial.print(" Pixel value written = ");Serial.println(wr,HEX);
  
  rd = tft.readPixel(30,30);
                Serial.print(" Pixel value read    = ");Serial.println(rd,HEX);

  // Walking 1 test
  wr = wr<<1;
  if (wr >= 0x10000) { while(1) yield(); };
}
```
Expected output:
```
 Pixel value written = 1
 Pixel value read    = 1
 Pixel value written = 2
 Pixel value read    = 2
 Pixel value written = 4
 Pixel value read    = 4
 Pixel value written = 8
 Pixel value read    = 8
 Pixel value written = 10
 Pixel value read    = 10
 Pixel value written = 20
 Pixel value read    = 20
 Pixel value written = 40
 Pixel value read    = 40
 Pixel value written = 80
 Pixel value read    = 80
 Pixel value written = 100
 Pixel value read    = 100
 Pixel value written = 200
 Pixel value read    = 200
 Pixel value written = 400
 Pixel value read    = 400
 Pixel value written = 800
 Pixel value read    = 800
 Pixel value written = 1000
 Pixel value read    = 1000
 Pixel value written = 2000
 Pixel value read    = 2000
 Pixel value written = 4000
 Pixel value read    = 4000
 Pixel value written = 8000
 Pixel value read    = 8000

```

Bodmer, 2021-02-12:
Ah, does your board use the new ESP32-S2 single core ESP32? If so then that would expalin the problem. The SDA single pin read may not work for that processor. You will need to use the older ESP32, the library has not been tested on the S2 single processing core variant!

sn00zerman, 2021-02-12:
Hi Bodmer,

*Stupid me !* I completely overlooked "It is possible to make the MOSI pin bidirectional at the ESP32 and ILI9488 so that the display can be read using readpixel ..." In your first response with the solution. (Sorry for taking up valuable time with my stupidness)
Thanks very much, all your examples do work !
I even use (already for a long time), Display SPI frequency = 55.00 - very stable !
Tomorrow I can start taking screenshots, just in time, as planned :-)

Thank you & best regards from Belgium,
Kris

Bodmer, 2021-02-12:
Great!

The ESP32 SPI cl;ock rates availble are integer divisions of 80MHz, so setting to 55MHz will end up being changed to 40MHz becuase the SPI library will use a lower available speed.

closed 2021-02-12 by Bodmer