Bodmer/TFT_eSPI #986: Clear text

epalzeolithe, 2021-02-06:
How i can easyly delete a string written with tft.drawString(), without flashing the screen (i fill a rectangle with background color before re-writing new string)

Thansk

Bodmer, 2021-02-06:
Use:
```
tft.setTextPadding(w);
```
Where w is the width of the text string in pixels. You can then print anything (including a "space" and it will over-write the shole line.

There are examples that show how this works.

closed 2021-02-06 by Bodmer

bastianhjaeger, 2022-05-09:
Note:
https://github.com/Bodmer/TFT_eSPI/blob/master/TFT_eSPI.h#L584
```
           setTextColor(uint16_t fgcolor, uint16_t bgcolor, bool bgfill = false),
```

has to be used as well with `bgfill = true`

(Took me some time to find that extra part)

Bodmer, 2022-05-10:
The extra parameter is only needed for the Smooth fonts. Further info in ReadMe:
https://github.com/Bodmer/TFT_eSPI#readme

DIYsciBorg, 2022-10-14:
If you set the text position using .setCursor, AND your background is a solid color (such as TFT_BLACK), you can reset the position back to that location, change the text color to the background, and "retype" the text, effectively erasing it.

`
gfx.fillScreen(TFT_BLACK);
gfx.setTextColor(TFT_WHITE);
gfx.setCursor(10, 50);
gfx.print("Hello!");
delay(5000);
gfx.setTextColor(TFT_BLACK);
gfx.setCursor(10, 50);
gfx.print("Hello!");
gfx.setTextColor(TFT_BLACK);
gfx.print("Good-bye!");
`