Bodmer/TFT_eSPI #987: Touch controller problems.

Mark-FS, 2021-02-06:
I can use the LCD and touch function with an ESP32 devkitC board just fine but not with a Heltec LoRa 32 v2 board.

If Lora 32 v2 (ESP32 based) I can use the LCD perfectly but the touch screen is not working correctly.  I have commoned the MISO, MOSI and CLK pins together for both the LCD and the touch controllers and I am using different CS pins.

On the LoRa 32 it does not allow me to calibrate the screen and rather than wait for a press it simply moves on to the next point after about half a second and repeats until it has finished.  I have unplugged the CS wire to the touch screen as this makes no differences and it still tries to calibrate itself.

I am using 23 and 17 for the CS for the LCD and the touch controllers.  I have reversed these and it makes no difference.  The LCD is happy either way and the touch is not happy.

I am a newbie and I hope I am not missing the obvious.  Any help would be much appreciated.  Thank you.

Mark-FS, 2021-02-06:
I just plugged in a version of the display with no touch screen controller soldered to the PCB, it also does the same and calibrates itself.  It all works fine on a normal ESP32 so there is something about the LoRa 32 v2 that is causing the calibration to fail.  Any ideas on a fix/workaround?  Cheers.

Mark-FS, 2021-02-06:
Looking a the other threads I tried the Read User Setup script and get

16:15:38.331 -> 
16:15:38.331 -> [code]
16:15:38.331 -> TFT_eSPI ver = 2.3.59
16:15:38.331 -> Processor    = ESP32
16:15:38.331 -> Frequency    = 240MHz
16:15:38.331 -> Transactions = Yes
16:15:38.331 -> Interface    = SPI
16:15:38.331 -> Display driver = 9341
16:15:38.331 -> Display width  = 240
16:15:38.378 -> Display height = 320
16:15:38.378 -> 
16:15:38.378 -> MOSI    = GPIO 27
16:15:38.378 -> MISO    = GPIO 19
16:15:38.378 -> SCK     = GPIO 5
16:15:38.378 -> TFT_CS   = GPIO 23
16:15:38.378 -> TFT_DC   = GPIO 13
16:15:38.378 -> TFT_RST  = GPIO 1
16:15:38.378 -> TOUCH_CS = GPIO 17
16:15:38.378 -> 
16:15:38.378 -> Font GLCD   loaded
16:15:38.378 -> Font 2      loaded
16:15:38.378 -> Font 4      loaded
16:15:38.378 -> Font 6      loaded
16:15:38.378 -> Font 7      loaded
16:15:38.378 -> Font 8      loaded
16:15:38.378 -> Smooth font enabled
16:15:38.378 -> 
16:15:38.378 -> Display SPI frequency = 27.00
16:15:38.378 -> Touch SPI frequency   = 2.50
16:15:38.378 -> [/code]

Test Touch Controller script runs but values never change wherever I touch the screen.  16:06:33.428 -> x: 1028     y: 1028     z: 4095

Bodmer, 2021-02-06:
Assuming it is wired up correctly then look for pin conflicts with other hardware that is on the board.

Maybe go back to using the devkit with those pins just to prove it can work.

Bodmer, 2021-02-06:
GPIO 1 is the serial output so generaaly that pin should be avoided. You are sharing the SPI bus with the LORA chip so make sure the Lora chip select pin is driven high. Maybe just connect the TFT_RST to  ESP32 RST?

Mark-FS, 2021-02-07:
Spot on, thank you Bodmer.

For the benefit of anyone else trying to use the LoRa 32 module, the CS for the LoRa controller is 18, defining it as an output and writing it high mutes LoRa and all works great.

You can move MOSI and MISO in the User_Setup.h file and it runs fine.  I moved them to 12 and 22 and that works without having to mute LoRa.  The downside to this is SPI speed is reduced from 80Mhz to 40Mhz for all SPI buses but I don't think that will affect me.

Lastly, connecting the LCD reset to GPIO 1 is a dumb idea and resets the LCD every time Serial.print is used.

Thanks again for the pointers.  Onwards and upwards....

Bodmer, 2021-02-07:
No problem, glad to help.

closed 2021-02-07 by Bodmer