Bodmer/TFT_eSPI #995: STM32F103C8T6 + 3.5inch SPI Module (ILI9488+XPT2046) How can I use the touchscreen?

lmdpua, 2021-02-10:
Hello. I have a module like [this](http://www.lcdwiki.com/3.5inch_SPI_Module_ILI9488_SKU:MSP3520).

[image]

The display works great with your library! How can I use a touchscreen driver? I have not found such a configuration

Bodmer, 2021-02-10:
See the [setup example here](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setups/Setup32_ILI9341_STM32F103.h).

QuitButton, 2021-02-11:
I'm just trying this exact display with a STM32F1 BluePill and I'm getting a lot less speed than the results here 
https://forum.arduino.cc/index.php?topic=655347.0

My results with SPI_FREQUENCY  55000000 and #define STM32:
`Benchmark                Time (microseconds)
Screen fill              421772
Text                     78870
Lines                    483421
Horiz/Vert Lines         44619
Rectangles (outline)     29625
Rectangles (filled)      868773
Circles (filled)         220613
Circles (outline)        292774
Triangles (outline)      112154
Triangles (filled)       397712
Rounded rects (outline)  135110
Rounded rects (filled)   935990
Done!
`

Whereas https://forum.arduino.cc/index.php?topic=655347.0 shows
`
Benchmark                Time (microseconds)
Screen fill              114895
Text                     21052
Lines                    139092
Horiz/Vert Lines         11158
Rectangles (outline)     7613
Rectangles (filled)      238863
Circles (filled)         80542
Circles (outline)        86984
Triangles (outline)      31905
Triangles (filled)       104423
Rounded rects (outline)  38057
Rounded rects (filled)   276499`

STM32F767 55MHz SPI, STM32 specific driver ILI9341
Benchmark                Time (microseconds)
Screen fill              114895
Text                     21052
Lines                    139092
Horiz/Vert Lines         11158
Rectangles (outline)     7613
Rectangles (filled)      238863
Circles (filled)         80542
Circles (outline)        86984
Triangles (outline)      31905
Triangles (filled)       104423
Rounded rects (outline)  38057
Rounded rects (filled)   276499

Bodmer, 2021-02-11:
@QuitButton There are a number of reasons for this:
1. Typically the benchmarks are for a n ILI9341 which is 320x240 pixels, the ILI9488 is 480*320 which is twice the number of pixels so the times double
2. The ILI9341 screen update uses 16 bits per pixel, the ILI9488 requires 24 bits per pixel so the times must be multiplies by 1.5
3. The Blue Pill is a 72MHz processor, so comparison with a faster 216MHz STM32F767ZI at 55MHz is not appropriate and without overclocking the maximumSPY rate for a Blue Pill is 36MHz

So taking the results from this post, you have to multiply the results x 2 and x 1.5 = x3 total.

Taking the results from that post, viz:
```
STM32 Blue Pill 36MHz SPI ILI9341
Benchmark,                Time (microseconds)
Screen fill,              377133
Text,                     70972
Lines,                    426499
Horiz/Vert Lines,         40546
Rectangles (outline),     26729
Rectangles (filled),      785933
Circles (filled),         257208
Circles (outline),        276474
Triangles (outline),      98671
Triangles (filled),       355996
Rounded rects (outline),  120036
Rounded rects (filled),   926630
Total = 3762827us
Total = 3.7628s
```
Would suggest that the graphics test will take ~ 10 seconds. In practice you may get a better result as some performance improvements have been made since that post.

lmdpua, 2021-02-12:
> See the [setup example here](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setups/Setup32_ILI9341_STM32F103.h).

Thanks for the quick answer. I managed to launch the touchscreen, but only separately. Both the display and the touchscreen do not want to work at the same time. I use example Test_Touch_Controller. Separately, the touch works, but as soon as I connect the SPI bus of the display, the touch stops working

Bodmer, 2021-02-12:
See the [warning here](https://github.com/Bodmer/TFT_eSPI/discussions/898).

In particular:

The ILI9488 SPI output ( SDO/MISO) does not go tristate, this cannot be circumvented, so **do not connect** the SDO/MISO pins on an ILI9488 if other SPI devices are on the same SPI bus.

lmdpua, 2021-02-12:
Ha! I unplugged (SDO / MISO) from the display and everything started working! Thank you!

closed 2021-02-12 by Bodmer