Bodmer/TFT_eSPI #1003: Wrong colors in screenshot

sn00zerman, 2021-02-12:
Hi Bodmer,

Following my previous issue, I have another issue.
Screenshot works fine, but image-colors are wrong.
I notices this issue in only on jpg-parts of the screen ...
Characters and solid colors work fine, it's only when there is a JPG in the screen itself.
I already lowered SPI speed to 20 Mhz and read speed to 5 Mhz, I also dit put a higher number in "PIXEL_TIMEOUT" in the screenServer.ino

I think it's not a communication issue, because the "artifacts" are consisted. (always at the same identical place)
This is how a screenshot looks like:
https://digitalplayground.be/tmp/screenshot_2021_2_12_18_38_2.png

On the screen itself, everything is crisp ! (it's hard to take a good picture of the screen)
The orange dots that form LOCOMOTION, are all fully filled, and the locomotive itself is more black in real life (at the backside)
also the start, power-icon, wifi-symbol and battery-symbol are crips in real life.

https://digitalplayground.be/tmp/foto_2021_2_12_19_6_1.png

Because it's hard to see how the real picture of the train looks like, here is a page with the same identical image and LOCOMOTION image used:
https://www.digitalplayground.be/locomotion.html

It actually looks like the pixels that are returned, do not contain enough bit-depth information ?

sn00zerman, 2021-02-12:
setting NPIXELS 1, solved my issue :-)

Bodmer, 2021-02-12:
Thanks for reporting this. I will add the bidirectional mode to the library. It seems I then have another problem to solve for block reads.  I will look into this before I add the new feature.

Bodmer, 2021-12-13:
Closing since I am unable to replicate the problem. Since there have been code iterations since this was raised the issue has probably been fixed.

closed 2021-12-13 by Bodmer