Bodmer/TFT_eSPI #999: Choppiness / Tearing when using 8 bit parallel display with esp32

j45p41, 2021-02-12:
Hi,

I am have posted this[Link](https://forum.lvgl.io/t/esp32-ili9488-8-bit-parallel-choppy-animation-when-scrolling-tabs-with-lvgl-generated-objects/4540)  issue on the lvgl forum. Given that I am using 8 bit parallel can you tell me if this issue is due to the way that parallel mode performance and if it increasing to 16 bit would resolve the issue. I would have thought that 8bit would be fast enough... thanks!

Bodmer, 2021-02-12:
The performance limits are due to a number of factors:

1. The ILI8488 requires 16 bits per pixel = 2 bytes
2. The ESP32 has a 20MHz I/O clock for register changes this means that a pin can only be changed every 50ns, which gives a maximum clock speed on a toggled pin of 10MHz = 5Mbytes/s or 80Mbps

The frame rate depends on the number of pixels that require updates, since you are updating almost the whole screen then the theoretical render time for a screen is 46.08ms. The TFT_eSPI library actually achieves 46.48ms so it manages 99.1% of the theoretical speed. Clearly this means the best I could do is increase performance by less than 1% which is insignificant visually.

In contrast the STM32F7 can toggle pins at 50MHz, i.e. 10 x faster. So the library actually has to slow down the toggling for that processor as this is too fast for the display. In practice the STM32F7 can update the screen 4x faster in (~12ms to update whole screen), but even this is strictly faster than the display datasheet maximum figures.

Moving the ESP32 to 16 bit would double the speed but this is not supported by the library and is not a change I would be willing to do. DMA is not supported by the library for parallel interfaces either athough it is possible to do this via the I2S bus feature of the ESP32, however this is not supported by the library either. DMA would only be worthwhile if LVGL is grabbing a significant amount of processor time and the best you could get is a 2x performance improvement for the optimum situation where the LVGL processing time is identical to the display update time. So you need to measure this time by some means to work out what performance gain could be achieved in  reality.

closed 2021-02-12 by Bodmer

j45p41, 2021-02-12:
@Bodmermany thanks for the response. I will go away and digest... However this might be a case of trading off certain visual effects against the benefit they bring.