Bodmer/TFT_eSPI #1007: TTGO Camera plus - config looks fine, still no example works and no error message

NRJMons, 2021-02-15:
Hi Bodmer

First of all, thank you for your work, it's really hard for a beginner to find his way around and I appreciate greatly your work.

After some reading, I finally understood what to do and uncommented the line Setup44 for my TTGO Camera plus in the User_Setup_Select.h, and I also commented the default User_setup.h define.

I then uploaded to my TTGO the example "Read user setup.ino" and got the following in the serial monitor:

[code]
13:50:25.111 -> TFT_eSPI ver = 2.3.59
13:50:25.111 -> Processor    = ESP32
13:50:25.111 -> Frequency    = 240MHz
13:50:25.111 -> Transactions = Yes
13:50:25.111 -> Interface    = SPI
13:50:25.111 -> Display driver = 7789
13:50:25.146 -> Display width  = 240
13:50:25.146 -> Display height = 240
13:50:25.146 -> 
13:50:25.146 -> MOSI    = GPIO 19
13:50:25.146 -> MISO    = GPIO 22
13:50:25.146 -> SCK     = GPIO 21
13:50:25.146 -> TFT_CS   = GPIO 12
13:50:25.146 -> TFT_DC   = GPIO 15
13:50:25.146 -> 
13:50:25.146 -> Font GLCD   loaded
13:50:25.146 -> Font 2      loaded
13:50:25.146 -> Font 4      loaded
13:50:25.146 -> Font 6      loaded
13:50:25.146 -> Font 7      loaded
13:50:25.146 -> Font 8      loaded
13:50:25.146 -> Smooth font enabled
13:50:25.146 -> 
13:50:25.146 -> Display SPI frequency = 40.00
13:50:25.146 -> 
[/code]

So, to me, it looks fine.

I then tried to upload the code "Colour test", no error, it boots fine.
But nothing happends, TFT screen remains black.

Can you please help me?

Big thanks in advance.

Nico

Bodmer, 2021-02-15:
Hi,
You seem to have done all the right things. The setup 44 was provided by a user as working but I note that no backlight pin has been declared. Is the display backlight on? The backlight is on GPIO 2 and can be controlled from you sketch.

At this point I would not fit an SD card as that can cause a conflict. When you do use a SD card then you will need to ensure the chip select is set high at the beginning of setup(), othrwise there may be aSPI bus conflict. simplest starting point though is not not use an SD card, then, providing the backlight is on the examples should run.

NRJMons, 2021-02-16:
Hey Bodmer

Thanks a lot, problem solved indeed, I modified Setup44 based on what I found in Setup12 for backlight variable:

#define ST7789_DRIVER

#define TFT_WIDTH  240
#define TFT_HEIGHT 240

#define TFT_MOSI    19
#define TFT_MISO    22
#define TFT_SCLK    21
#define TFT_CS      12
#define TFT_DC      15
#define TFT_RST     -1
**#define TFT_BL	   2 // Backlight PIN ADDED here by me**

My screen works now perfectly, all the examples work but are displayed as rotated 90° to the right...strange
But it's probably the examples that are generic, I'll figure it out.

Thanks again for the fantastic work AND speed of reaction.

Nico

PS: I guess you need also to modify Setup44 in your source code now, sorry I don't know how to do that. ;)

NRJMons, 2021-02-16:
I think you can close the issue, problem is solved

closed 2021-02-16 by NRJMons

Bodmer, 2021-02-16:
I have updated setup 44, thanks for reporting this.