Bodmer/TFT_eSPI #1009: Use more recent TFT_eSPI with M5Stack Core 2 under PlatformIO?

mikeaich, 2021-02-16:
Hi, I'm running into a number of build issues trying to convince PlatformIO to use a more recent version of TFT_eSPI than the one included in the M5Stack Core 2 library.

My `platformio.ini' file looks like this:
```
[env:m5stack-core-esp32]
platform = espressif32
board = m5stack-core-esp32
framework = arduino
lib_deps = 
    https://github.com/Bodmer/TFT_eSPI.git
    https://github.com/m5stack/M5Core2.git
upload_speed = 921600
upload_port = /dev/ttyUSB0
monitor_port = /dev/ttyUSB0
monitor_speed = 115200
build_flags =
    -DBOARD_HAS_PSRAM
    -mfix-esp32-psram-cache-issue
```
...and in my `main.cpp`, I start with:
```
#include <M5Core2.h>
#include "TFT_eSPI.h"
```
Furthermore, I have modified `TFT_eSPI/User_Setup_Select.h` to comment out `User_Setup.h` on line 22 and enable `User_Setups/Setup12_M5Stack.h` on line 36, but all I've managed to achieve is more build errors than the VSCode terminal can capture.

(I'm hoping a newer version of TFT_eSPI will solve some PSRAM-sprite pushing lockups I'm seeing. Thanks in advance and thank you for putting together such a great library!)

Bodmer, 2021-02-16:
The M5Stack team have made quite a few changes to the version they embedded. I am not able to support products that use altered variants of my library.

closed 2021-02-16 by Bodmer

mikeaich, 2021-02-16:
Yeah, I realized that when I was trying to merge your latest version into a clone of their repo.

Digging some more, I actually think I'm running into issue https://github.com/Bodmer/TFT_eSPI/issues/734 (I'm trying to _slide_ a full-screen JPEG in from the side and seeing lots of tearing, lock-ups, incomplete images). I'll try again with smaller sprites, and if that doesn't work, I might take a stab at ripping out their stuff and adapting any changes to your library.