Bodmer/TFT_eSPI #1016: Display not working on esp8266

fungreenfox, 2021-02-22:
Trying to setup this display with TFT_eSPI:
http://www.lcdwiki.com/3.5inch_Arduino_Display-UNO?fbclid=IwAR1-e-5H3zdsiPYh-XwdURdHv27Lhrmk8jdAuMJWFjkVeJeLQDmAPAzaG4c#Product_Picture
It does not turn on.
Tried different settings, among this one 
#define TFT_SDA 13   //
#define TFT_MISO 12 //  SD0/MISO#
//#define TFT_MOSI 13 //13 //
#define TFT_SCK 14   //
#define TFT_CS 0 // Chip select control pin D8
#define TFT_DC 16 // Data Command control pin
#define TFT_RST -1  // Set TFT_RST to -1 if the display RESET is connected to NodeMCU RST or 3.3V

Bodmer, 2021-02-22:
That is a parallel only display, parallel displays with the ESP8266 are not supported due to lack of pins.

It will work with an ESP 32 UNO board.

closed 2021-02-22 by Bodmer

fungreenfox, 2021-02-22:
I have a Wroom also.

Bodmer, 2021-02-23:
I suggest an ESP32 UNO board with the mods detailed in the ReadMe to maximise the chance of success.

fungreenfox, 2021-02-23:
This Wroom is also an esp32, with a lot of pins. Its just about if the display runs parralel on this?

Bodmer, 2021-02-23:
Yes, parallel is supported for ESP32.

fungreenfox, 2021-02-23:
Im not sure i got all setup right. The 9486 script has been chosen. compiling says FT_eSPI\Extensions/Touch.cpp:104:3: error: 'spi' was not declared in this scope
   spi.transfer(0xb0);               // Start new Z1 conversion
   ^
exit status 1
Error compiling for board ESPea32.

Bodmer, 2021-02-23:
Do not specify the TOUCH_CS pin.