Bodmer/TFT_eSPI #1018: Can a JPG be pushed out in a sprite

stephensaid, 2021-02-22:
Can I push a JPG to a Sprite instead of outputting it right after the sprite?

Bodmer, 2021-02-23:
Yes, see the [example here](https://github.com/Bodmer/TFT_eSPI/tree/master/examples/Sprite/Rotated_Sprite_3).

closed 2021-02-23 by Bodmer

stephensaid, 2021-03-02:
Thanks, Bodmer.

I've implemented TFT_eFEX.

I now have a problem with the images. The following image is not right.
[image]

It should look like:
[image]

I have other images with the same behaviour.

I've tried changing tft.setSwapBytes(true); to tft.setSwapBytes(false); but it has no effect.

Please help.