Bodmer/TFT_eSPI #1022: fillSprite AND WHITE COLOR

toosimit, 2021-02-23:
HI
 I have problem with fillSprite when I set color to white other color no problem when I set color to white draw a circle  what can I do 
core is esp8266 and compiler is Arduino IDE

`// Created by Bodmer 17/3/20 as an example to the TFT_eSPI library:
// https://github.com/Bodmer/TFT_eSPI

#define NEEDLE_LENGTH 25  // Visible length
#define NEEDLE_RADIUS 80  // Radius at tip

#include <TFT_eSPI.h>
TFT_eSPI tft = TFT_eSPI();
TFT_eSprite needle = TFT_eSprite(&tft); // Sprite object for needle

// =======================================================================================
// Setup
// =======================================================================================
void setup()   {
  Serial.begin(115200); // Debug only

  tft.begin();
  tft.setRotation(3);
  tft.fillScreen(TFT_WHITE );
//tft.setSwapBytes(true);
  // Create the needle Sprite
  createNeedle();

  // Define where the needle pivot point is on the TFT
  tft.setPivot(tft.width()/2, tft.height()/2);  // Set the TFT pivot point

  // Full scale to sweep area (angle, delay_per_degree)
  plotNeedle(359, 1);

  // Reset needle position to 0
  plotNeedle(0, 1);
}

// =======================================================================================
// Loop
// =======================================================================================
void loop() {
  // Plot needle at random angle in range 0 to 240, speed 5ms per degree
  plotNeedle(random(360), 5);

  // Pause at new position
  delay(250);
}

// =======================================================================================
// Create the needle Sprite
// =======================================================================================
void createNeedle(void)
{
  needle.setColorDepth(8);
  needle.createSprite(5, NEEDLE_LENGTH);  // create the needle Sprite 11 pixels wide by 30 high

  needle.fillSprite(TFT_WHITE); // Fill with black

  // Define needle pivot point
  uint16_t piv_x = needle.width() / 2;   // pivot x in Sprite (middle)
  uint16_t piv_y = NEEDLE_RADIUS;        // pivot y in Sprite
  needle.setPivot(piv_x, piv_y);         // Set pivot point in this Sprite

  // Draw the red needle with a yellow tip
  // Keep needle tip 1 pixel inside dial circle to avoid leaving stray pixels
//  needle.fillRect(piv_x - 1, 2, 3, NEEDLE_LENGTH, TFT_RED);
  needle.fillRect(piv_x - 1, 2, 3, 5, TFT_GREEN);
}

// =======================================================================================
// Move the needle to a new position
// =======================================================================================
void plotNeedle(int angle, byte ms_delay)
{
  static int16_t old_angle = 0; // Starts at -120 degrees

  if (angle < 0) angle = 0; // Limit angle to emulate needle end stops
  if (angle > 361) angle = 359;

  angle -= 0; // Starts at -120 degrees

  // Move the needle until new angle reached
  while (angle != old_angle) {

    if (old_angle < angle) old_angle++;
    else old_angle--;

    // Draw the needle in the new postion
    needle.pushRotated(old_angle);

    // Slow needle down slightly as it approaches new postion
    if (abs(old_angle - angle) < 10) ms_delay += ms_delay / 5;

    // Wait before next update
    delay(ms_delay);
  }
}

// =======================================================================================`
[image]
for that

Bodmer, 2021-02-23:
This is caused by a bug and this has been fixed in the github master now.

Thanks for reporting this.

I have updated you sketch. I assume you wish the background to be cleared to black, to do this the needle sprtie needs to be wider with a black border to avoid leaving stray pixels behind, this is needed as the 1 degree increments move the needle about 2 pixels at a time.

Updated sketch:

```
// Created by Bodmer 17/3/20 as an example to the TFT_eSPI library:
// https://github.com/Bodmer/TFT_eSPI

#define NEEDLE_LENGTH 25 // Visible length
#define NEEDLE_RADIUS 80 // Radius at tip

#include <TFT_eSPI.h>
TFT_eSPI tft = TFT_eSPI();
TFT_eSprite needle = TFT_eSprite(&tft); // Sprite object for needle

// =======================================================================================
// Setup
// =======================================================================================
void setup() {
  Serial.begin(115200); // Debug only

  tft.begin();
  tft.setRotation(3);
  tft.fillScreen(TFT_BLACK );
  //tft.setSwapBytes(true);
  // Create the needle Sprite
  createNeedle();

  // Define where the needle pivot point is on the TFT
  tft.setPivot(tft.width() / 2, tft.height() / 2); // Set the TFT pivot point

  // Full scale to sweep area (angle, delay_per_degree)
  plotNeedle(359, 1);

  // Reset needle position to 0
  plotNeedle(0, 1);
}

// =======================================================================================
// Loop
// =======================================================================================
void loop() {
  // Plot needle at random angle in range 0 to 240, speed 5ms per degree
  plotNeedle(random(360), 5);

  // Pause at new position
  delay(250);
}

// =======================================================================================
// Create the needle Sprite
// =======================================================================================
void createNeedle(void)
{
  needle.setColorDepth(8);
  needle.createSprite(9, NEEDLE_LENGTH); // create the needle Sprite 11 pixels wide by 30 high
  needle.fillSprite(TFT_BLACK);

  // Define needle pivot point
  uint16_t piv_x = needle.width() / 2; // pivot x in Sprite (middle)
  uint16_t piv_y = NEEDLE_RADIUS; // pivot y in Sprite
  needle.setPivot(piv_x, piv_y); // Set pivot point in this Sprite

  // Draw the red needle with a yellow tip
  // Keep needle tip 2 pixel inside dial circle to avoid leaving stray pixels
  // needle.fillRect(piv_x - 1, 2, 3, NEEDLE_LENGTH, TFT_RED);
  needle.fillRect(piv_x - 2, 2, 5, NEEDLE_LENGTH - 3, TFT_WHITE);
  needle.fillRect(piv_x - 1, 2, 3, 5, TFT_GREEN);
  //needle.pushSprite(0,0);
}

// =======================================================================================
// Move the needle to a new position
// =======================================================================================
void plotNeedle(int angle, byte ms_delay)
{
  static int16_t old_angle = 0; // Starts at -120 degrees

  if (angle < 0) angle = 0; // Limit angle to emulate needle end stops
  if (angle > 361) angle = 359;

  angle -= 0; // Starts at -120 degrees

  // Move the needle until new angle reached
  while (angle != old_angle) {

    if (old_angle < angle) old_angle++;
    else old_angle--;

    // Draw the needle in the new postion
    needle.pushRotated(old_angle);

    // Slow needle down slightly as it approaches new postion
    if (abs(old_angle - angle) < 10) ms_delay += ms_delay / 5;

    // Wait before next update
    delay(ms_delay);

  }
}

// =======================================================================================
```

closed 2021-02-23 by Bodmer

toosimit, 2021-02-24:
THANK YOU SO MUCH 
I WANT TO ADD TWO QUESTIONS
1-HOW CAN ADD 2 TIPS NEEDLE LIKE COMPASS NEEDLE?
2-HOW CAN CHANGE TIP OF NEEDLE TO Triangle INSTEAD OF RECTANGLE?

commit 2bf4d15b (2021-02-23): Fix #1022