Bodmer/TFT_eSPI #1034: GC9A01 Animated_dial needle "trailing shadow " issue

flykarlos, 2021-03-01:
Hi Bodmer
I would like to report issue with example Animated_dial on GC9A01 display. Needle is leaving Black shadow efectively clearing dial behind it when it moves. Is the TFT CGRAM not accessible on GC9A01? If yes, is there any workaround?
Thanks!

https://user-images.githubusercontent.com/18057422/109553521-6b311900-7ad3-11eb-8557-9914ff7eef69.mov

Bodmer, 2021-03-02:
Assuming you are using the small circular screen available from China, then my tests indcate the screen cannot be read. I suspect the display board has hardwired the IM(x) pins into the wrong state.  The display has a single SDA pin (no MOSI+MISO pins) so if it could be read then the library must be configured using  **_#define TFT_SDA_READ_** in the setup file. As this appears to be a limitation of the way the display is built, then nothing can fix this in the library (unless there is a command that I have missed after reading through the datasheet!).

If your ESP32 has PSRAM then a fulls creen 240x240 sprite could be created and used as a frame buffer.

The GC9A01 support is still in the "experimental" phase and thus is currently not listed as a supported device in the ReadMe. I received one of these displays a few days ago and can confirm that sprite plotting with a transparent colour (used for needle in you sketch) does not work properly. I noticed this with the Orrery sketch, however that sketch works fine on other dsplays, so there is something different about in the way the GC9A01 must be driven compared to other displays. This is under investigation.

Bodmer, 2021-03-02:
See also #867

flykarlos, 2021-03-03:
Thanks for explanation. I tried to enable `#define TFT_SDA_READ`, but effect was that display was flickering once in every few seconds with strange artifacts and then it was slowly fading out. So `#define TFT_SDA_READ` is clearly not applicable to this Chinese display using GC9A01 driver. 
Waiting for any news regarding Your investigation about this issue. Once again thanks for Your hard work.

serdargoksal, 2021-05-06:
According to my experience you can read screen with readRect(), when you uncomment 
#define  USE_HSPI_PORT in user_setup.h
However in this case drawLine(), drawText() does not work.

Bodmer, 2021-11-28:
I have investigated this and it appears that some GC9A01 screens cannot be read.

closed 2021-11-28 by Bodmer

zweann, 2022-05-14:
Hi Bodmer,
i tried two different GC9A01 displays, but both can't be read. It's so annoying!
Can you provide a link list with working GC9A01 round displays?

RickJHarris, 2022-09-03:
> Assuming you are using the small circular screen available from China, then my tests indcate the screen cannot be read. I suspect the display board has hardwired the IM(x) pins into the wrong state. [...]

I've been trying to get the Animated_dial to work properly on a GC9A01. If I define TFT_SDA_READ do I need to define an SDA pin and connect it to the  GC9A01? If so, which pin? Thanks!

01GOD, 2023-09-08:
Sigh...saw now. There are white square graphics artifacts that follow the indicator around. I'm visiting shenzhen now (not chinese) and hoping that gets solved. By the way, the adafruit library example DOES display properly. So that is some hope! 

I think bodmer will get it solved soon.

bluecraank, 2024-11-07:
Any news on this?

I bought the waveshare gc9a01 round display and it have cgram of about 192,000 bytes.

Reading the display is not working.

Tried everthing mentioned here.