Bodmer/TFT_eSPI #1035: Printing Decimal or Float crashes ESP32 - How to do this?

dookie454, 2021-03-02:
Hello, I've tried two of the most common ways to print a decimal, and both crash the ESP32.

I've tried making a float, then simply printing that, but it crashes right away, before anything gets printed to the screen.

Also tried the trick of printing and integer with the original value mult by 10,  then in the print request, divide by 10, and add 1.  
Closest i can get is for example, 9,7 to show as 97,  with no decimal.  

Please help  if you can !!  :)   

tft.drawString
tft.drawNumber
tft.print(message_E85, 1);   This gets the closest, it shows a 0.0 for about 1 sec, but then ESP32 crashes. 
tft.print(message_E85/10, 1);   This gets the closest, it shows a 0.0 for about 1 sec, but then ESP32 crashes.

dookie454, 2021-03-02:
UPDATE:  I found solution somewhere else, that works great.  Using "tft.drawFloat". 

float E85_Decimal = message_E85;       //https://forum.arduino.cc/index.php?topic=544537.0
tft.drawFloat(E85_Decimal/10, 1, 80, 222, 1);  // Need to have the OBD message mult by 10 for this to work.

Bodmer, 2021-03-02:
It is unlikely the crash is due to the floating point value. tft.print() cause the number to be processed by the Arduino core steam function and that is unlikely to have a bug.  Check if the ESP32 watchdog is timing out, or you are running off the end of an array elsewhere in your code causing variables to be corrupted. To stop a watchdog timeout include yield(): in you sketch if a function is going to take some time (>1 second) to execute.

closed 2021-03-02 by Bodmer