Bodmer/TFT_eSPI #1038: Image problems using TFT_eFEX library

stephensaid, 2021-03-02:
Thanks, Bodmer.

I've implemented TFT_eFEX.

I now have a problem with the images. The following image is not right.
[image]

It should look like:
[image]

I have other images with the same behaviour.

I've tried changing tft.setSwapBytes(true); to tft.setSwapBytes(false); but it has no effect.

Please help.

_Originally posted by @stephensaid in https://github.com/Bodmer/TFT_eSPI/issues/1018#issuecomment-789177553_

Bodmer, 2021-03-03:
I don't have enough information to help. You need to provide a complete sketch that illustrates the problem and describe your hardware setup.

fotosettore, 2021-03-10:
hi
the same for me ...
the sketch is the demo : Rotated_Sprite_3 and image is the original eye.
thanks for help!