Bodmer/TFT_eSPI #1039: Crash when rendering fonts when allocating TFT_eSPI using 'new'

Shadetheartist, 2021-03-02:
Firstly I'll admit that my C++ experience is limited, so this may be a silly question. After several hours of debugging my code thinking that there must be an issue on my end, I have ended up isolating the issue down to this library, specifically relating to how the TFT_eSPI object is created. The board I'm using is a TTGO T-Display, which is an ESP32 & ST7789, with a 240x135px ips display.

The following code works correctly, as in the examples.

```
#include <TFT_eSPI.h> // Hardware-specific library
#include <SPI.h>

// Invoke custom library
TFT_eSPI tft = TFT_eSPI();

void setup(void) {
  Serial.begin(115200);

  tft.init();
  tft.setTextWrap(false);
  tft.fillScreen(TFT_BLACK);
  tft.setCursor(0, 30);
  tft.setTextColor(TFT_RED);
  tft.setTextSize(1);
  tft.println("Hello World!");
}

void loop() {
  
}

```

The following code does not work. The only difference being that the tft object is allocated on the heap, as far as I know.

```
#include <TFT_eSPI.h> // Hardware-specific library
#include <SPI.h>

// Invoke custom library
TFT_eSPI *tft = new TFT_eSPI();

void setup(void) {
  Serial.begin(115200);

  tft->init();
  tft->setTextWrap(false);
  tft->fillScreen(TFT_BLACK);
  tft->setCursor(0, 30);
  tft->setTextColor(TFT_RED);
  tft->setTextSize(1);
  tft->println("Hello World!");
}

void loop() {
  
}

```

I am getting `Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.` when I try to render strings using print, drawString, drawChar etc.

The stack trace is always pointing toward the Smooth_font.cpp file.

```
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d20d4  PS      : 0x00060530  A0      : 0x800d2545  A1      : 0x3ffb1ea0  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x0000f800  A5      : 0x00000048  
A6      : 0x00000001  A7      : 0x0000f800  A8      : 0x00000000  A9      : 0x1ab38c35  
A10     : 0x00000000  A11     : 0x3ffba478  A12     : 0x3ff65000  A13     : 0x00000000  
A14     : 0x00060720  A15     : 0x00000000  SAR     : 0x0000001c  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x1ab38c3d  LBEG    : 0x400e8408  LEND    : 0x400e846a  LCOUNT  : 0x00000003  

ELF file SHA256: 0000000000000000

Backtrace: 0x400d20d4:0x3ffb1ea0 0x400d2542:0x3ffb1f00 0x400ee3c3:0x3ffb1f60 0x400d0e82:0x3ffb1f80 0x400d4886:0x3ffb1fb0 0x4008620d:0x3ffb1fd0
  #0  0x400d20d4:0x3ffb1ea0 in TFT_eSPI::drawChar(int, int, unsigned short, unsigned int, unsigned int, unsigned char) at .pio/libdeps/esp32dev/TFT_eSPI/Extensions/Smooth_font.cpp:519
  #1  0x400d2542:0x3ffb1f00 in TFT_eSPI::drawChar(unsigned short, int, int, unsigned char) at .pio/libdeps/esp32dev/TFT_eSPI/Extensions/Smooth_font.cpp:519
  #2  0x400ee3c3:0x3ffb1f60 in TFT_eSPI::drawChar(unsigned short, int, int) at .pio/libdeps/esp32dev/TFT_eSPI/Extensions/Smooth_font.cpp:519
  #3  0x400d0e82:0x3ffb1f80 in setup() at src/main.cpp:34
```

Shadetheartist, 2021-03-02:
I should note that you can circumvent this issue to some degree via:
```
TFT_eSPI _tft = TFT_eSPI();
TFT_eSPI *tft = &_tft;
```
I guess I'm just wondering if there's some limitation of the library that I've missed, or if it's undocumented.

Bodmer, 2021-03-03:
I assume ther error message is for a different sketch to the short example you provide as it should not be generating an error associated with smooth fonts...

Unfortunately the error is a limitation of the ESP32 environment and supporting core package (not the TFT_eSPI library).  If you run sketches that do not directly read the FLASH memory (this is where font bitmaps are stored) then you will probably find those sketches work fine.

Incidentally, the code sketch you provide (that generates an error on an ESP32) would run fine on an STM32 board.

closed 2021-03-03 by Bodmer

Shadetheartist, 2021-03-03:
> I assume ther error message is for a different sketch to the short example you provide as it should not be generating an error associated with smooth fonts...

That's why I'm so confused, the example code I showed before was indeed generating an error, the only difference being that the error was for drawChar and the code is using println. I double-checked to make sure. 

This code:
```
#include <TFT_eSPI.h> // Hardware-specific library
#include <SPI.h>

// Invoke custom library
TFT_eSPI *tft = new TFT_eSPI();

void setup(void) {
  Serial.begin(115200);

  tft->init();
  tft->setTextWrap(false);
  tft->fillScreen(TFT_BLACK);
  tft->setCursor(0, 30);
  tft->setTextColor(TFT_RED);
  tft->setTextSize(1);
  tft->println("Hello World!");
}

void loop() {
  
}

```

 
Generates this error:
```
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d2f7c  PS      : 0x00060530  A0      : 0x800ee678  A1      : 0x3ffb1ef0  
A2      : 0x3ffba478  A3      : 0x00000048  A4      : 0x00000000  A5      : 0x00000048  
A6      : 0x00060720  A7      : 0x00000000  A8      : 0x1b338c34  A9      : 0x3ffb1ed0  
A10     : 0x00000048  A11     : 0x00000048  A12     : 0x3ffb8488  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x3ffb0060  SAR     : 0x0000001c  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x1b338c3e  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffc  

ELF file SHA256: 0000000000000000

Backtrace: 0x400d2f7c:0x3ffb1ef0 0x400ee675:0x3ffb1f20 0x400d37fe:0x3ffb1f40 0x400d3825:0x3ffb1f60 0x400d0e81:0x3ffb1f80 0x400d48b2:0x3ffb1fb0 0x4008620d:0x3ffb1fd0
  #0  0x400d2f7c:0x3ffb1ef0 in TFT_eSPI::write(unsigned char) at .pio/libdeps/esp32dev/TFT_eSPI/Extensions/Smooth_font.cpp:519
  #1  0x400ee675:0x3ffb1f20 in Print::write(unsigned char const*, unsigned int) at /home/derek/.platformio/packages/framework-arduinoespressif32/cores/esp32/Print.cpp:90
  #2  0x400d37fe:0x3ffb1f40 in Print::write(char const*) at /home/derek/.platformio/packages/framework-arduinoespressif32/cores/esp32/Print.cpp:90
  #3  0x400d3825:0x3ffb1f60 in Print::print(char const*) at /home/derek/.platformio/packages/framework-arduinoespressif32/cores/esp32/Print.cpp:90
      (inlined by) Print::println(char const*) at /home/derek/.platformio/packages/framework-arduinoespressif32/cores/esp32/Print.cpp:195
  #4  0x400d0e81:0x3ffb1f80 in setup() at src/main.cpp:16
  #5  0x400d48b2:0x3ffb1fb0 in loopTask(void*) at /home/derek/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:32
  #6  0x4008620d:0x3ffb1fd0 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

Rebooting...
```

If it's not an issue with the library, could you please point me in the right direction to understand the issue with the ESP32? I have been battling this issue for like two days now and still don't know why it's occurring. So any help would be greatly appreciated.

Bodmer, 2021-03-03:
I think you will need to dig dee into the ESP32 core package and make some significant changes to fix this. You may need to dig deep into the ESP IDF as well.  Start by searching though the core "[issues](https://github.com/espressif/arduino-esp32/issues)" as I anticipate this problem has been seen before by someone else.

Personnally I would use the appoach that you know works and focus on completing your project based on the library examples rather than being distracted by a problem that only occurs in very particular circumstances.

Shadetheartist, 2021-03-03:
Thanks for the advice! As you say, I will probably just ignore the underlying issue for now, since I have a workaround.