Bodmer/TFT_eSPI #1044: Problem with 4" tft arduino shield display

adrianiancu, 2021-03-04:
Hello, My name is Adrian and I am not able to make a 4" tft display (arduino shield, SKU 13587 - waveshare) working. I was using another display using ILI9341 and I had no problem. I modified the User_Setup.h for ILI9486 and my pin connections but... 
Any suggestion? 
Thank you,
Adrian.

adrianiancu, 2021-03-04:
I forgot to mention that the display is functional with Arduino uno and Waveshare_4_Inch_Tft_Touchscreen library.

Bodmer, 2021-03-05:
The Waveshare screen you have is fitted with a 16 bit serial shift register. If this is the same circuit as the Waveshare Raspberry Pi displays then the TFT_eSPI setup option **_#define RPI_DISPLAY_TYPE_** may work. See [setup 11](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setups/Setup11_RPi_touch_ILI9486.h).

Bodmer, 2021-03-13:
Assumed resolved by user.

closed 2021-03-13 by Bodmer

adrianiancu, 2021-03-19:
I apologies for the delay. I had to take care of my mom in the hospital. Thank you a lot for your answer. It works. I appreciate the energy, time and passion you invest in this.
Adrian