Bodmer/TFT_eSPI #1046: ESP32 TTGO T-Display blank screen with DMA

Riffer, 2021-03-04:
I use the Sprite Cube example with Setting 25 for the T-Display. 

It compiles and works very well without DMA, nice demo.

With DMA the screen stays blank, without DMS I see the cube dancing and in both cases, I get FPS from serial debug. I played around with the CPU load and it turns out with DMA the CPU has much more power left.

I already tried with only 27 Mhz for SPI but did not help. 

Is this display even capable of DNS? Maybe I bark up the wrong tree?

Bodmer, 2021-03-05:
I have tried DMA with the TTGO T Display v1.1 ([product detail here](http://www.lilygo.cn/prod_view.aspx?TypeId=50033&Id=1126&FId=t3:50033:3)) and DMA works fine with [setup 25](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setups/Setup25_TTGO_T_Display.h) and the SpriteRotating Cube example. You will find that 80MHz will not work with DMA for the SPI clock rate due to the pins used for SPI and limitiations within the ESP32 hardware, so 40MHz is the maximum for DMA. Frame rate will be ~160fps with DMA and ~130fps without DMA at 40MHz SPI rate.

My display seems to work at 80MHz SPI rate when DMA is **_NOT_** used and then the frames rate is ~220fps.

Riffer, 2021-03-05:
Hello @Bodmer 

Thank you very much for your response and the clarification. My TTGO has printed 19-6-28 V1.1 XY-CP on the silk - so this should be the same as yours.

I tried different settings with platformio again and turned to the basic Arduino platform, and nothing helped.
I have two devices on the shelf, and I tried the other one but did not succeed with DMA.

Maybe this is just a knockoff board or something, but thanks again for your response! I do not depend on DMA - it was just to test how it works - and in my case, it doesn't.

Bodmer, 2021-03-06:
Those silk screen markings are identical to mine, so it is not clear to me why my board works with DMA and yours does not. Which version of the TFT_eSPI library are you using?

Riffer, 2021-03-07:
For Platform IO:
bodmer/TFT_eSPI @ 2.3.60

Arduino - just do make sure I use the latest verson - I downloaded the ZIP copied the contenct to the library folder. Then usedthe example from the GUI menu. I only changed User_Setup_Select.h to to use 25.

With DMA the screen has background light but the cube only appears when comment the define for DMA out.

With DMA off I have aprox 130 fps and with DMA on (blank screen) 155 fps.

Bodmer, 2021-03-10:
Which version of the ESP32 core board support package do you use?

Riffer, 2021-03-11:
I tried 1.0.5 first and then 1.0.4 in Arduino IDE.

At platformio I am back to Arduino ESP 32 3.0.0 (which includes 1.0.4) because I had strange problems with another project that showed same symptoms with 1.0.5 (not related to DMA) on Arduino IDE.

Bodmer, 2021-03-13:
OK, I have tried 1.0.4 and 1.0.5 and DMA works OK, so unfortunately I have no idea why it does not work for you.

closed 2021-03-13 by Bodmer

Bodmer, 2021-03-13:
OK, I noticed a subtle difference in my setup file that appeared incosequential but was not!

The issue reported is now fixed at version 2.3.61

Thanks for reporting this bug.

Riffer, 2021-03-14:
Thank you very much! I confirm this bug is fixed, too. Updated to 2.3.61 and compiled with 1.0.4 with Arduino IDE.

Bodmer, 2021-03-14:
Thanks for confirming.

stevesch, 2021-05-22:
Excellent!  working for me as well w/2.3.69

commit cd025dbc (2021-03-13): Fix #1046

ESP32 DMA did not work if MISO pin is not defined in setup file (affected TTGO T-Display setup 21)