Bodmer/TFT_eSPI #1049: Setup36_RPi_touch_ILI9341.h missing

Andy-45, 2021-03-05:
This file in User_Setups missing:
//#include <User_Setups/Setup36_RPi_touch_ILI9341.h>      // Setup file configured for ESP32 and RPi TFT with touch

seems to be replaced with Setup36_RPi_touch_ST7796.h

could you please add it again

Bodmer, 2021-03-06:
I have corrected the User_Setup_Select.h file to reference the correct setup 36 file name. Thanks for pointing this out.

As far as I am aware the ILI9341 Raspberry Pi TFT board does not work with the library. Reason is unknown as I am unable to test, thus it would be inappropriate to have a setup file for the ILI9341 Raspberry Pi TFT. If you have a working setup then let me know and I will add your setup file.

Andy-45, 2021-03-06:
the ILI9341 / XPT2046 (Arduino UNO Shield ) is working fine now with https://github.com/prenticedavid/MCUFRIEND_kbv 
thank you!

closed 2021-03-07 by Bodmer