Bodmer/TFT_eSPI #1052: Arduino due custom font support.

EliasA97, 2021-03-06:
Using arduino due I can't compile the fonts examples. I get a ('SPIFFS' was not declared in this scope) error.

Bodmer, 2021-03-06:
The Due does not support SPIFFS (ESP32 and ESP8266 board do). However there should be other examples that use arrays and do work OK.

Exactly which examples are you trying to run?

EliasA97, 2021-03-06:
Hi Bodmer, 

I found out that using the arrays example I can use .vlw fonts. I sense that you don't have much time for answering everyone's questions, so I will make you 4 quick questions in order not to take much of your time.

1) How can I generate and use non smoothed fonts?
2) Using an ILI9488 on arduino Due, is it faster at 8bit parallel or hardware SPI? Also do you have any info about speed 
    comparison between your library and UTFT?
3) Is DMA supported on the Due?
4) Is there any manual for your library?

Thank's for all your help!

Bodmer, 2021-03-06:
1. The library supports Adafruit_GFX format fonts, there are web based font generators on the web (easy to Google) or for windows this [command line executable](https://github.com/Bodmer/GFX_Font_Converter) can be used.
2. DUE is likley to be faster with this library but I have not tried it. As the DUE will be treated as a generic processor there is no dramatic optimisation like the STM32, ESP32 and ESP8266 code.
3. No
4. A manual is on the TODO list, but usually I find something more interesting to do! There is [some info here](https://github.com/Bodmer/User_Manual_TFT_eSPI) on functions. There are plenty of examples that demonstrate most features. It is Adafruit_GFX compatible to an extent but may have deviated over the last 4 years from full mcompatibility, however their tutorials will help.

closed 2021-03-06 by Bodmer

EliasA97, 2021-03-07:
Hi Bodmer, just a little update:

I compared the ili9488 SPI using your library with the same tft but in 16bit parallel mode using UTFT. UTFT works faster. I can't find any way to use your library with 8bit parallel conection though.