Bodmer/TFT_eSPI #1054: Rotation and Inversion Problem.

JinLuckyboy, 2021-03-06:
``` c++
uint16_t calData[5] = { 0, 1023, 0, 1023, TOUCH_SCREEN_INFORMATION };
tft.setTouch(calData);
```

getTouch() is depending to TOUCH_SCREEN_INFORMATION only.
Draw methods are depending to setRotation().
Therefore, a parameter of setRotation() that does not same rotation set when I invoke calibrateTouch(), makes "touch error about rotation and inversion".

``` c++
tft.rotation(0);
uint16_t calData[5];
tft.calibrateTouch(calData, TFT_MAGENTA, TFT_BLACK, 15);
tft.setTouch(calData);
// no error

tft.rotation(1);
// error!

tft.rotation(2);
// error too!
```

I think getTouch() is depending to setRotation(), too.

Bodmer, 2021-03-06:
The calibration must be done in the screen rotation you wish to use. Alternatively there are other independant libraries [such as this one](https://github.com/achillhasler/TFT_eTouch) that are very good.

The touch support will be deprecated eventually as it makes sense to have independant libraries.

closed 2021-03-06 by Bodmer

JinLuckyboy, 2021-03-07:
All of other libraries do not work my device(ILI9341).
So, I made some code, and share it.

``` c++
// define function
void setRotation(uint8_t rotation) {
  // binary invert_y invert_x rotate
  // rotation_index[] = {rotation[0].calDate[4], rotation[1].calDate[4], ...}
  uint16_t const rotation_index[] = {0b110, 0b011, 0b000, 0b101};//{6, 3, 0, 5}
  uint16_t const calData[4] = { 443, 3471, 244, 3565 }; // calData excluding index 4
  uint8_t const calRotation = 0; // rotation when you calibrate bringing calData

  if(!(0 <= rotation && rotation <= 3)) return;

  if(rotation%2 == calRotation%2) {
    uint16_t parameters[] = {calData[0], calData[1], calData[2], calData[3], rotation_index[rotation]};
    tft.setTouch(parameters);
  } else {
    uint16_t parameters[] = {calData[2], calData[3], calData[0], calData[1], rotation_index[rotation]};
    tft.setTouch(parameters);
  }

  tft.setRotation(rotation);
}
```

``` c++
// use
setRotation(0); // or setRotation(1); or ...
// you don't need to invoke setTouch()
```