Bodmer/TFT_eSPI #1056: Another flovour of ILI9481 or ILI9486

RalfJL, 2021-03-07:
Hi,
I bought this Display:
https://de.aliexpress.com/item/1005001989363608.html?spm=a2g0s.9042311.0.0.5e7b4c4dmeLrsL
The IPS/Touch version which claims to have a ILI9481. 
The IPS version is looking, so much better then the old ones.
But:
I tried all 6 Versions of the ILI9481 init but none worked good. The RGB was looking the best, but still a lot of issues.
In my desperation I tried the ILI9486 setup and that looked very good, beside the fact, that everything is mirrored.
changing 0x36,0x48 to 0x36,0x8c did the job. Now everything looks good.
So what is it? A new flavour of ILI9481 or a new flavour of ILI9486. I decided to add a new flavour to the ILI9481 init, because it has already some choices.

Unfortunately the display seems to be electrically unstable or it is still not the right setup.
It is attached to a ESP32 DevKit that also does the power supply. The rainbow sketch has some flickering in its colors so have buttons or any non black areas, and the display turns dark from time to time. Sometime approaching the display with the hand is enough to turn the display dark.  I attached a low resistant capacitor between VDD and GND, but that did not really help. I also added a pullup resistor to RST but no luck. Still unstable.

At the moment I can't recommend that display.
Any help is appreciated.
Thanks
Ralf

RalfJL, 2021-03-07:
Another observation,

reading from the Display holds the display, no update but still the old content visible.
log_d("Read from TFT: %d", tft.readPixel(0, 0));

Bodmer, 2021-03-07:
I had a quick look at the vendors page and notice that they provide 3 controller options. You may have been sent the wrong one?

The photo's also indicate **_NO_** 5V to 3.3V regulator is fitted! So **_only power te display from 3.3V_** if your display also does not have the regulator fitted. The data is misleading as if suggests the input is 5V or 3.3V but that depends on the parts fitted.

The ILI9481/86 appear to be "older" tech and do not support the overclocking more revent displays support. The data sheet maximu is 10MHz so anything over that is overclocking. Some displays can be overclocked by 2x to 4x the data sheet figures. I found 16MHz worked for my ILI9481 but not 20MHz, so 12MHz is probably safe from the reliability viewpoint.

For the 480x320 display options the ST7796 is probably the best for speed and run as 40MHz well, mine seems to run at 80MHz reliably but others have noticed errors at that speed. However the easily available displays have a hardware design bug, [see here](https://github.com/Bodmer/TFT_eSPI/discussions/898).

The[ MHS 4.0 Raspberry Pi display based on the ST7796](http://www.lcdwiki.com/MHS-4.0inch_Display-B) controller runs reliably at 80MHz on an ESP32 (40MHz on ESP8266).

RalfJL, 2021-03-07:
Hi,
thanks for the answer and thanks for the good work.
The display is definitely 3.3v. It has a regulator (small one upper left corner), but as the vendor shows there is a resistor R0 that has a "0" on it, I assume it is a bridge. The regulator is inactive until you cut out the 0 Ohm resistor.
And I am pretty sure that it is a IPS panel. I have several other older TFT 2.8" and 3.5" (ILI9341 based) and their colors fade when you are not looking straight to it.  The colors of the one I bought are very stable no matter what angle you are looking at it. Actually this is the reason I bought it, I wanted an IPS panel. 
That does not necessarily mean that the controller really is a ILI9481.
Currently the most annoying problem is it's sensitivity to electrostatic. It is still winter and the air is dry, so when I sit down I always get electrocuted when I touch something that is grounded like the radiator or my computer mouse.
After I dischargeing myself I can touch the case or the display without any problem. As soon as I move on the chair a bit the display turns off, when I am closer than 5 cm to the display.
In my eyes the display has a poor electrical design and, as you mentioned, an old controller. My older TFT displays do not have that problem. By the way, I bought 2 of them and both show the same problems.

I wasted now 3 days with it and I am running out of ideas and patients. 
Best is to follow your recommendation and buy a display with a better controller. 

Thanks very much for your time. You may close that thread, if you like.

Ralf

closed 2021-03-09 by Bodmer

RalfJL, 2021-03-09:
Well, I did take all the anti static precautions. I am working a lot with electronic  and computer parts.
But it all does not help. Mounting the ESP32 and the TFT into a 3D printed case, the display turns black even if you approach the housing. You don't have to touch the housing or the display (housing is made of PLA).
For a touch display not an option.
On a breadboard it is more stable. It is a bit further away from the ESP32 antenna which has WiFi always on. 

The jack is exactly the same as for my old 2.8 Displays and actually you can't do anything wrong here, beside swapping MOSI and MISO.

You might be right that I might have a flaw in some where.

On last question. When I disconnect 3.3v from the display only, I would expect that the display would have to be re initiated, when I reattach 3.3v, right?
That does not happen. The back light turns on again and the display is working normally. I think that is odd and may point to my flaw.

What do you think?

RalfJL, 2021-03-15:
Solution.
i figured it out.
The WiFi antenna of the ESP32 Dev Board was very close to the displays connector ( about 1 centimetre). That obviously caused interference with the SPI signals. Approaching the side with the antenna with the hand obviously increased the interference. 
So the Display was unreliable and that caused the confusion about the nature of it.

I changed my 3D printed case and used cables to move the ESP32 board away from the display (around 8 centimetres)  and that fixed the issue.
So I am running the display now as:
- IL9481
- 10MHz SPI speed
- RGB flavour
No artefacts, no crashes
A wonderful clear and color stable Display now.

Thanks again
Ralf