Bodmer/TFT_eSPI #1064: Compatibility TFT LCD 3.5 "SPI ili9488 touch

Jorge1998D, 2021-03-11:
Hi,
I spent several hours looking for a 3.5 "touch LCD and only found this: 3.5" TFT LCD SPI ili9488 touch.
[image]
Is it compatible with these libraries? And if so, who do I connect the touch pins to? because I investigated your documentation and I could not find it.
Thanks for your time.

dookie454, 2021-03-11:
I have a 2.8" board with the same pins, this is what works for me, and yes, documentation is not clear about this.  

For ESP32 Dev board (only tested with ILI9341 display)
  Updated by Bodmer 21/12/16 for TFT_eSPI library:
  https://github.com/Bodmer/TFT_eSPI
The hardware SPI can be mapped to any pins, these are what I currently use:
#define TFT_MISO 19  //SDD on LCD AND T_DO (for touch)
#define TFT_MOSI 23  //SDI on LCD AND T-DIN (for touch)
#define TFT_SCLK 18  //SCK on LCD AND T-CLK (for touch)
#define TFT_CS   15  //CS on LCD - Chip select control pin - AND T_CS (For touch)
#define TFT_DC    2  //DC on LCD - Data Command control pin
#define TFT_RST   27  //4  RESET pin on LCD -  Reset pin (could connect to RST pin)   Devkitv1 does not have pin 0!
#define T_CS   21 // Touch Control Pin

Jump these to the existing PINs above
T_DO -> TFT_MISO pin
T_DIN -> TFT_MOSI pin
T_CLK -> TFT_SCLK pin
T_CS -> ---> D21 on ESP32

copyCATandpasteDOG, 2021-03-12:
Got this one and use it with wemos mini d1,here's my config:

CS=D8 , DC=D3 , MOSI & T_DIN=D7
  SCK & T_CLK=D5 , MISO=NOTHING!
  T_CS=D4 , T_DO=D6 , LED=5V
  RESET=3,3V

I use it with ILI9488 and STA7796 too with this config

Bodmer, 2021-03-13:
Yes, that display will work with touch. Note that the picture poseted does not have U2 the XPT2046 fitted, however if you order a display with the touch screen then it will be fitted.

The touch controller and TFT must be wired to share the SPI bus, the exception for the ILI9488 is that you MUST NOT wire the SDO(MISO) pin to the processor as it does not go tristate (if you do wire it the touch will not work, [see alert](https://github.com/Bodmer/TFT_eSPI/discussions/898)).

Pins connections:
```
#define TFT_MISO 19    // T_DO (for touch)
#define TFT_MOSI 23    // SDI on LCD AND T_DIN (for touch)
#define TFT_SCLK 18    // SCK on LCD AND T_CLK (for touch)
#define TFT_CS 15      // CS on LCD - Chip select control pin
#define TFT_DC 2       // DC/RS on LCD - Data Command control pin
#define TFT_RST 27     // RESET pin on LCD - Reset pin (could connect to processor RST pin)
#define TOUCH_CS 21    // Touch Control Pin T_CS
```

The T_IRQ pin is unused.

The display is fitted with a 5V to 3.3V regulator so can be powered with 5V to VCC and 0V to GND

closed 2021-03-13 by Bodmer