Bodmer/TFT_eSPI #1065: Cannot write to display control registers in ili9488

monith202, 2021-03-11:
I am using ESP32 to interface with 320 *480 lcd driven by ILI9488.I am able to read all the default register values through SPI Interface.But I can't write any values to the display control registers like memory access control or the frame memory write.They all show the default values after writing to those registers.The display always stays white.But on doing SLPOUT and DISPON it just dims a little.Can you please help solve this issue?

Bodmer, 2021-03-13:
Do you have the display working and running the examples? If so post your sketch so I can see what you are trying to do.

monith202, 2021-03-13:
Yes.All this time I've been using the VSPI pins with the MISO Line connected.When I changed the user setup and connected to HSPI instead examples run fine even with the MISO connected.I don't know why MISO is causing this.But anyway it works.Thanks for responding to my problem.

Bodmer, 2021-03-13:
The TFT MISO line on the ILI9488 displays will causes issues if other SPI peripherals share the bus. In that case leave it unconnected.

Glad to here you have it working.

closed 2021-03-13 by Bodmer