Bodmer/TFT_eSPI #1067: GC9A01 Rotated_Sprite_3.ino pixels of random colors- compression error?

flykarlos, 2021-03-13:
First of all:
Thanks @Bodmer  for this wonderful library and tremendous amount of work You are putting into maintenance of it!

I started using Your library to drive GC9A01 display. Everything works fine, except rotation of sprite with jpeg plotted on it. I thought that there is error in my code, so I reverted back to examples and flashed unchanged Rotated_Sprite_3.ino example. Result is that some image is plotted on the screen. There is some similarity to image stored in data folder, however looks like each end every pixel has random color. Just like some kind of compression error. Have a look:

[image]
[image]

Any help will be appreciated.
Processor: ESP32 ESP-WROOM-32D
environment: MacOS Mojave, Arduino IDE 1.8.13
Of course I am using ESP32FS tool to upload sketch data as described in example.
Thanks!

closed 2021-03-13 by Bodmer

markotime, 2022-07-19:
Have put .jpgs into every SPIFF directory I can find, but the arduino serial monitor reports that no files found.  This includes the SPIFF subdir enumerated during compilation.  Any help greatly appreciated!

Galaxy-Man, 2022-08-19:
Morning all? I have trie dthe code and uploaded the compass.jpg file off of this page. When runnign the code it says 

Initialisation done.

Listing SPIFFS files:
=================================================
  File name                              Size
=================================================
  /compass.jpg                          17126 bytes
=================================================

Jpeg file format not supported!

Any ideas??

Galaxy-Man, 2022-08-19:
Further info. 
I right clicked on the origial jpeg: above and saved, then copied into the \data directory then used the comannd line to upload to the ESP.