Bodmer/TFT_eSPI #1070: 8 bits ILI9225 screen

pierbout, 2021-03-13:
Hello,

Your library works fine on ILI9225 TFT screens with hardware SPI.

Now, I'd like to use it with a 8 bits ILI9225 screen (so 8 distinct wires) on a 3.3v Mega2560 MEGA PRO MINI (because this screen is thinner and cheaper).

I tried in User_Setup.h:
            #define TFT_CS   A11  // = SS : Chip select control pin
            #define TFT_DC   A12  // = RS : Data Command control pin
            #define TFT_WR   A13  //4  // Write strobe control pin
            #define TFT_RD   38 // VCC : Read strobe control pin (constant 3.3v)
            #define TFT_RST  A14  // Reset pin, toggles on startup
            #define TFT_D0   39
            #define TFT_D1   40
            #define TFT_D2   41
            #define TFT_D3   42
            #define TFT_D4   43
            #define TFT_D5   44
            #define TFT_D6   45
            #define TFT_D7   46

Note: LED  is connected to GND

The screen leds swich on but nothing is displayed

Any idea how to declare it correctly?

Furthermore, do you have an idea how to make sSPI detect that it is connected to a _bits screen and not an hardware SPI?

Thank you in advance for your help

[image]

Bodmer, 2021-03-13:
The library supports [STM32 and ESP32 processers](https://github.com/Bodmer/TFT_eSPI#8-bit-parallel-support) in 8 bit parallel mode, not generic Arduino boards like the Mega. Support could be added but the performance would be worse than a SPI interface unless optimisation was added for a particular processor.

Either stick with SPI or try the MCUfriend_kbv library which is good and supports the ILI9225. The author also provides excellent support on the Arduino forum for the library.

The more advanced features of the TFT_eSPI library use a lot of RAM and thus you will only be able to achieve the same functionality level as with the [MCUfriend_kbv library](https://github.com/prenticedavid/MCUFRIEND_kbv) and due to the optimisation in that library dedicated to the Mega it will be much faster for rendering to the screen.

closed 2021-03-13 by Bodmer

pierbout, 2021-03-13:
Thank you Bodmer. I'm going to follow your ardvice.