Bodmer/TFT_eSPI #1077: Touch coordinates and display rotation

daxnv, 2021-03-15:
the coordinates you get from getTouch(&x,&y) only fit to drawPixel(x,y) if rotation is 1.

Bodmer, 2021-03-15:
You need to run the calibration function on the rotation you wish to use, [see sketch](https://github.com/Bodmer/TFT_eSPI/blob/master/examples/Generic/Touch_calibrate/Touch_calibrate.ino#L22).

The 5th parameter in the calibration array is a set of binary flags that can be used in a sketch to mirror/swap x and y individually.  

Adding auto rotation to the library is possible but are 8 ways  (x,y swap = 2 and mirror options =4) these touch screen can be fitted and wired. Different suppliers use different orientations on seemingly identical looking screens, so this makes it more complicated than it might appear.

closed 2021-03-15 by Bodmer

Bodmer, 2021-03-15:
I have updated the sketch coment to clarify. Thanks for reporting this, I can see it might be a source of confusion.