Bodmer/TFT_eSPI #1080: Touch working, but nothing on screen.

holmlrs, 2021-03-17:
Hi! Thanks for maintaining this nice library!

I have an issue I have not encountered before. On a custom ESP board with an ESP pico D4, I attached a 2.8" TFT screen with an ili9341 controller and touch. All the connections mentioned below have been verified.

This is my config file:

```
#define ILI9341_DRIVER

#define TFT_MISO 12 //MTDI
#define TFT_MOSI 13 
#define TFT_SCLK 14
#define TFT_CS   15  //MTDO Chip select control pin
#define TFT_DC    8 // Data Command control pin
#define TFT_RST   2  // Reset pin (could connect to RST pin)
#define TOUCH_CS 4     // Chip select pin (T_CS) of touch screen
//#define TFT_RST  -1  // Set TFT_RST to -1 if display RESET is connected to ESP32 board RST

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:-.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
//#define LOAD_FONT8N // Font 8. Alternative to Font 8 above, slightly narrower, so 3 digits fit a 160 pixel TFT
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

#define SMOOTH_FONT

//define SPI_FREQUENCY   1000000
//#define SPI_FREQUENCY   5000000
//#define SPI_FREQUENCY  10000000
 //#define SPI_FREQUENCY  20000000
#define SPI_FREQUENCY  27000000
//#define SPI_FREQUENCY  40000000

//#define SPI_READ_FREQUENCY  20000000

#define SPI_TOUCH_FREQUENCY  2500000

#define USE_HSPI_PORT

 //#define SUPPORT_TRANSACTIONS

```

The problem is that the touch works flawlessly, but the screen does not. Uploading the Test_Touch_Controller sketch turns the screen black and reports correct ADC data on the serial port. This tells me that the SPI bus (MOSI,MISO,SCK) is functioning well at 25Mhz. Holding the reset button makes the screen go white. 

Now when I use the same config file for the TFT_graphicstest_PDQ sketch, it does not work as expected. 

- DC and RST show activity during initialisation
- CS,MISO, MOSI,CLK show activity during test.
- The screen uses a 3V3 supply, the jumper to bypass the LDO is bridged.
- (un)commenting HSPI port does not change anything
- Lowering SPI freq changes nothing
- The pins in the config are confirmed connected correctly on the pcb.

However, the two video's below show the result on the screen Both videos show execution of the graphicstest:

https://user-images.githubusercontent.com/43642819/111477971-ecbdb380-872f-11eb-89fb-624fae1c1a26.mp4

https://user-images.githubusercontent.com/43642819/111477979-efb8a400-872f-11eb-9c72-e187db746c6b.mp4

Does anyone recognize this problem?

holmlrs, 2021-03-17:
Resolved, GPIO 8 turns out to be an input only pin, which was not documented in the pico-d4 datasheet. Switched to gpio23 and all is fine

Bodmer, 2021-03-17:
Glad you have this fixed and working.

To quote the data sheet:

"Pins IO16, IO17, CMD, CLK, SD0 and SD1 are used to connect the embedded flash, and can not be used for other purposes."

CMD = GPIO 11, CLK = GPIO 6, SD0 = GPIO 7 and SD1 = GPIO 8 so all these must be avoided.

closed 2021-03-17 by Bodmer