Bodmer/TFT_eSPI #1082: WaveShare Display not working

3mc2, 2021-03-17:
Hello,

I hope you can help.  I am using a Waveshare SPI Display 4" Display, with an ESP32.   Specifically https://www.amazon.com/gp/product/B07B9XPQS4/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1
All that I get is a white screen.   
It says it used the ILI9486 Driver, I have tried that and ILI9488 driver.  No Dice.  I have tried with RPI set on and off - No Dice

I have tried both the HSPI and VSPI pins on the ESP32, I have adjusted the config file in every manner I could come up with.

Any guidance you can provide would be greatly appreciated.

My current configuration is:
#define RPI_DISPLAY_TYPE - Have tried both enabled/disabled
#define ILI9486_DRIVER
#define TFT_MISO 19
#define TFT_MOSI 23
#define TFT_SCLK 18
#define TFT_CS   15 // Chip select control pin
#define TFT_DC    2  // Data Command control pin
#define TFT_RST   4  // Reset pin (could connect to RST pin)
//efine TFT_RST  -1  // Set TFT_RST to -1 if display RESET is connected to ESP32 board RST

#define TOUCH_CS 22    
 #define SPI_FREQUENCY  40000000
//#define SPI_FREQUENCY  27000000 Tried this one as well

Thanks

Rich

Bodmer, 2021-03-17:
These shift register based designs vary and unfortunately this means it is difficult to cater for every possible circuit.

The circuit design typically only works over a narrow range of SPI clock rates (typically 10 to 20MHz) as the design uses a logic "race hazard" which is poor design practice.

Try other SPI clock rates, it is unlikely to work at 40MHz.

These displays can be troublesome to get working so really are best avoided.

Bodmer, 2021-03-19:
See #1044 where a user appears to have the same display and has it working.

3mc2, 2021-03-23:
Thank you, for the assistance.  Can you recommend a display or 2 that work
well?

Rich

Bodmer, 2021-03-23:
The user is #1044 got i working. Have you managed this too?

The ILI9488 screens are common and work well, otherwise I lie the ST7796 screens but see #898.

closed 2021-03-27 by Bodmer