Bodmer/TFT_eSPI #1083: st7735 high brightness

gionni2010, 2021-03-17:
Hello, i'm using your library with a st7735 and it is working very well. so i bought 3 new screens for new peojects but they seems to have very high brightness, the screen is completely white, and you can see the text if you look ad the screen very near and not from the front (sorry, i'm not good with engliash and i can't explain well).

If i try the adafruit graphicstest sketch from the Adafruit_ST7735.h is workging good. I've tried every possible combinations on the User_Setup.h without luck.

This is the working tft screen: https://www.amazon.it/dp/B07N6FQ5XW/

This is the new NOT WORKING screen: https://www.amazon.it/gp/product/B07BY2W1TQ/

Can you help me please?

gionni2010, 2021-03-17:
I forget to say that i'm using them on an esp32

gionni2010, 2021-03-17:
[image]

This is the result (not working) with TFT_eSPI and the digital clock example

gionni2010, 2021-03-17:
[image]
And this is with adafruit_st7735 library

Bodmer, 2021-03-17:
The advert for the display that does NOT work says it has an SSD1306 controller and the TFT_eSPI library does not support this. The library does support the ST7735 controller in it's various configurations.

If you are convinced the advert is wrong, and the supplier provided a ST7735 type display by mistake then you could try the following:

1. Check your setup is being picked up by the compiler by running the diagnostic "Read_User_Setup" sketch and check the serial output.

2. Try a lower SPI clock rate.

3. Try a different "colour tab" option, [see here](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setup.h#L85-L106).

gionni2010, 2021-03-17:
Hello thank you for your fast reply!

I tryed all the color tabs and clock rates in the setup file with no luck.
And yes i think that on amazon there is wrong advert, I'll attach a pdf from the vendor showing right info. Like said before, those screens can run fine with the Adafruit_ST7735 library.

this is the console result from the "read_user_setup" sketch:

[code]
TFT_eSPI ver = 2.3.61
Processor    = ESP32
Frequency    = 240MHz
Transactions = Yes
Interface    = SPI
Display driver = 7735
Display width  = 128
Display height = 160

R0 x offset = 2
R0 y offset = 1
MOSI    = GPIO 23
SCK     = GPIO 18
TFT_CS   = GPIO 17
TFT_DC   = GPIO 2
TFT_RST  = GPIO 12

Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled

Display SPI frequency = 27.00
[/code]

gionni2010, 2021-03-17:
[1,8 TFT Display EN.pdf](https://github.com/Bodmer/TFT_eSPI/files/6160399/1.8.TFT.Display.EN.pdf)
Here the info from the vendor

Bodmer, 2021-03-18:
The connections and setup look OK. Since you already have one ST7735 display running fine this suggests there is something significantly different about the others.  Unfortunately there are an increasing number of "clone copy chips" coming out of China that have a lower quality and specification (they may even be marked to look like the original manufacturers chips!). This has happened to displays.

The Adafruit library is probably using a lower SPI clock rate. try reducing the rate with TFT_eSPI to 8MHz:

#define SPI_FREQUENCY  8000000

Another option is to try and buy more of the displays that you do have working.  Unfortunately I have run out of ideas otherwise.

gionni2010, 2021-03-18:
Thank you very much Bodmer, i sent them back :)

closed 2021-03-18 by Bodmer