Bodmer/TFT_eSPI #1090: help ili9488

Roman26rus, 2021-03-20:
hello. help please I use esp32 I connect the LCD via the parallel interface, I connect the touchscreen via SPI, how to use the touch screen via SPI code the mode of inclusion is parallel ? The ILI9488 LCD controller .touch screen XPT2046 .the problem is that the LCD only works on a parallel 8bit interface ,and the touch is connected to SPI,can you somehow make it work with the built-in library of the touch screen?

Pablo2048, 2021-03-20:
https://github.com/Bodmer/TFT_eSPI#touch-controller-support - read the first sentence in the paragraph...

Roman26rus, 2021-03-20:
[image]

Roman26rus, 2021-03-20:
well, in fact, you can make it work, but I tried the library <XPT2046_Touchscreen.h> it works in conjunction with TFT_eSPI
just I would like to just enable SPI reading of the touch screen in the display connection mode via a parallel bus

Pablo2048, 2021-03-20:
There is clearly written "The XPT2046 touch screen controller is supported for SPI based displays only." There was many discussions about this (did you search the closed issues?), for example here https://github.com/Bodmer/TFT_eSPI/issues/979#issuecomment-772109636 , https://github.com/Bodmer/TFT_eSPI/discussions/895 ...

Bodmer, 2021-03-20:
As noted by others above. #895 gives example code.

closed 2021-03-20 by Bodmer