Bodmer/TFT_eSPI #1094: pushImage at least bpp4 can't work with images in PROGMEM esp8266

ESPboy-edu, 2021-03-21:
using  
void pushImage(int32_t x, int32_t y, int32_t w, int32_t h, uint8_t  *data, bool bpp8 = true, uint16_t *cmap = nullptr);
for 4bbp as
tft.pushImage(16, 16, 16, 16, (uint8_t *)tile, false, bit4palette);
causes reset

but with sprite work ok
spr.createSprite(16, 16);
spr.createPalette(bit4palette); 
spr.pushImage(0, 0, 16, 16, (uint16_t *)tile);
spr.pushSprite(16, 16);

it seemed that pushImage at least bpp4 can't work with images in PROGMEM esp8266

_Originally posted by @ESPboy-edu in https://github.com/Bodmer/TFT_eSPI/discussions/1093_

Bodmer, 2021-03-21:
Yes, 4bpp from FLASH is not supported. The ESP8266 requires specific commands to access FLASH data, but I suspect it would work on an ESP32. 4bpp support was added by a library user (Kate Morris) but I have not used the feature except to test the addition.

Post a complete example sketch and I will see if it can be added easily.

ESPboy-edu, 2021-03-22:
[Tileset-map.zip](https://github.com/Bodmer/TFT_eSPI/files/6184789/Tileset-map.zip)

with line 59 uncommented - myESPboy.tft.pushImage(...) -- resets
with line 59 commented and lines 60,61 uncommented spr.pushImage(...) and spr.pushSprite(...) works fine

but if i will put all images to RAM (remove PROGMEM from images arrays) myESPboy.tft.pushImage(...) works fine also

Bodmer, 2021-03-23:
I do not have a Adafruit_MCP23017 or Adafruit_MCP4725, can you provide a simplified, minimal sketch that illustrates the problem. I can then use it to test any changes to the library.

ESPboy-edu, 2021-03-23:
here it is
[Tiles_4bit.zip](https://github.com/Bodmer/TFT_eSPI/files/6191623/Tiles_4bit.zip)

Bodmer, 2021-03-23:
Thanks I will have a look and see what needs to be done.

Bodmer, 2021-04-06:
I have updated the library.

I will have to find a better solution to duplicating functions but that will have to wait.

closed 2021-04-06 by Bodmer

Bodmer, 2021-04-06:
Forgot to mention, it is best to use
```
const uint8_t *
```
for FLASH pinters to invoke the right function. In the example this means:
```
tft.pushImage(i*16, j*16, WIDTH_TILE, HEIGHT_TILE, (const uint8_t *)tilesList[table[i][j]], false, bit4palette);
```

commit fd99db43 (2021-04-06): Fix #1094

Allow ESP8266 and RP2040 to use FLASH for pushImage () and 4bpp