Bodmer/TFT_eSPI #1096: Display freeze (ESP32/ST7335 1.8'')

Strg-Felix, 2021-03-22:
Hi,

the display just freezes after working once. The values on the TFT remain the same but LED and everything else still working correctly. Any ideas??

 static unsigned long lastEventTime = millis();
    static const unsigned long EVENT_INTERVAL_MS = 30000;

      if ((millis() - lastEventTime) > EVENT_INTERVAL_MS) {
bme.begin(0x76);
 digitalWrite(ledred, HIGH);
              
    
              tft.setCursor(0, 0);
              tft.fillScreen(TFT_BLACK);
              delay(3000);
              tft.setTextColor(TFT_WHITE);
              tft.setCursor(0,3);
              tft.setTextSize(2);
              tft.println(("DATA"));
              tft.setTextSize(2);
              tft.print("Temp.:");
              tft.println(bme.readTemperature());
              tft.print("Moist.:");
              tft.println(soils[0].getMoisture());
              tft.print("Humid.:");
              tft.println(bme.readHumidity());
              digitalWrite(ledred, LOW);
              delay(1);
lastEventTime = millis();

  }

Bodmer, 2021-03-22:
There is probably a conflict somewhere for pins used. Try simplifying the sketch by commenting out lines to see which line is the problem.

For example this will wok on any board:
```
#include <TFT_eSPI.h>
#include <SPI.h>

TFT_eSPI tft = TFT_eSPI();
void setup() {
  tft.init();
  tft.fillScreen(TFT_BLACK);
}

void loop() {
  static unsigned long lastEventTime = millis();
  static const unsigned long EVENT_INTERVAL_MS = 3000;

  if ((millis() - lastEventTime) > EVENT_INTERVAL_MS) {
    //bme.begin(0x76);
    //digitalWrite(ledred, HIGH);
    tft.setCursor(0, 0);
    tft.fillScreen(TFT_BLACK);
    delay(3000);
    tft.setTextColor(TFT_WHITE);
    tft.setCursor(0, 3);
    tft.setTextSize(2);
    tft.println(("DATA"));
    tft.setTextSize(2);
    tft.print("Temp.:");
    tft.println(2.345); //bme.readTemperature());
    tft.print("Moist.:");
    tft.println(6.789); //soils[0].getMoisture());
    tft.print("Humid.:");
    tft.println(55); //bme.readHumidity());
    //digitalWrite(ledred, LOW);
    delay(1);
    lastEventTime = millis();
  }
}
```

Bodmer, 2021-03-23:
Note that typically you would only run bme.begin(0x76); once is the setup().

Strg-Felix, 2021-03-23:
thx for your help. actually i commented all of your suggested lines out and added them partwise in again and now it works. kind of weird but as long as it works haha

Bodmer, 2021-03-23:
Glad to hear it.

closed 2021-03-23 by Bodmer