Bodmer/TFT_eSPI #1097: Issue in configuring User_setup

AamirJafari, 2021-03-22:
HI brother,
I am using ST7789 IPS display with 240 x240 pixels with esp32 board
while I am trying to setup the User_Setup file maybe I do something wrong or my display is broken and i got the display shown below.
https://prnt.sc/10spug3
Help me to find the problem is it faulty display or i wrong configured the file.

Bodmer, 2021-03-22:
Run the diagnostic sketch "Read_User_Setup" to check the compiler is picking up the settings you expect.

AamirJafari, 2021-03-23:
On using the above mentioned sketch my serial monitor did not show anything

AamirJafari, 2021-03-23:
Hi bro,
Can you please suggest me a video tutorial to setup my esp 32 for my 1.3"
240 x 240 display so I can test that my display is broken or not and if
broken I proceed my refund accordingly.
Please try to reply as soon as possible
Thank you

AamirJafari, 2021-03-23:
Thank you so much I found the problem.It is a loose connection and a micro mistake that I made in User_Setup file.
Now the library is working great.
Thanks Again

closed 2021-03-23 by AamirJafari

Bodmer, 2021-03-23:
Glad to hear it.