Bodmer/TFT_eSPI #1099: Show custom image on st7789

AamirJafari, 2021-03-24:
Hi there,
I am beginner and recently sorted out to use your library with my ESP32 and ST7789 240 x 240 display.
I want to display my image in the display without sd card but unable to find it out.Is there any example in the library to show image if yes then then tell me about the example else tell me how do I perform this task

Bodmer, 2021-03-24:
The answer will depend on the format of the image, e.g. jpeg, bmp (6/16/24 bits), raw bitmap of some kind etc. etc.

AamirJafari, 2021-03-25:
Actually I am working on a smartwatch project and I just want to show some images as my watchfaces so any format will work great that maintain quality and have less size.

Bodmer, 2021-03-27:
Try TFT_Flash_Bitmap (images need to be converted, see comments in sketch) or TFT_SPIFFS_BMP which draws a 16 bit bmp format image, this uses SPIFFS so you will need to make sure this is available in the IDE for uploading the file.

Bodmer, 2021-03-27:
Flash_Jpg_DMA should also work.

closed 2021-03-29 by Bodmer