Bodmer/TFT_eSPI #1100: Q: STM32F407 + SSD1963

dronecz, 2021-03-24:
Hi @Bodmer , 

I bought [this](https://www.aliexpress.com/item/32890948123.html) screen and because I do not like stock FW, I would like to write my own in Arduino IDE. I can change stock FW but there are some limits (especially price of IAR workbench) but before I start, I would like to confirm myself that this is doable (at least in HW point of view). I´m just hobby programmer so maybe some stuff will be wrong. 

1) I can [see](https://github.com/makerbase-mks/MKS-TFT-Hardware/blob/master/MKS%20TFT70/MKS%20TFT70%20V1.1_004/MKS%20TFT70%20V1.1_004%20SCH.pdf) that LCD is connected via 16-bit bus, but if I got info for this library right, you support just 8-bit parallel, is that right? I assume that in 8-bit setup should LCD work as well. 
2) I can not use touch driver with parallel connection so I have to use separate library for this, is that right?

Thanks for reply.

Bodmer, 2021-03-24:
1. Correct, 8 bit only.
2. Correct, a nother library is needed so suit the touch controller installed on the screen.

The errors look like you have multiple copies of the library installed. Look at the errors and see where the conflicting files are located.

If you can, run the Read_User_Setup example to and post the settings report picked up by the compiler.

dronecz, 2021-03-24:
I did find some issues with more than one instances of library so I update and deleted old TFT_eSPI lib folder before I edited my post with my porblem. I have no other folder with library in my folder: 

[image]

I tried to compile recomended sktech but result is the same. Here is log from compilation: 
[tft_espi_read_user_setup_log.txt](https://github.com/Bodmer/TFT_eSPI/files/6201087/tft_espi_read_user_setup_log.txt)

Thanks for your help.

Bodmer, 2021-03-26:
You appear to have renamed the library "arduino_565604", the library name should be TFT_eSPI.

Download and install via the IDE menu if you are having problems installing libraries.

Bodmer, 2021-03-27:
I have checked and the renaming of the library is not important, I can see a problem in the processor file TFT_eSPI_STM32.h

I will investigate the fix.

dronecz, 2021-03-27:
> You appear to have renamed the library "arduino_565604", the library name should be TFT_eSPI.
> 
> Download and install via the IDE menu if you are having problems installing libraries.

This name comes from library manager. I install all libraries via library manager if possible. 

> I have checked and the renaming of the library is not important, I can see a problem in the processor file TFT_eSPI_STM32.h
> 
> I will investigate the fix.

Ok, thank you very much!

Bodmer, 2021-03-27:
I have updated the [main repository](https://github.com/Bodmer/TFT_eSPI), you will need to download from Github (not using IDE as it has not been released yet). I have updated TFT_eSPI_STM32.h

I see the code was marked as "untested" so please report any problems as I do not have a suitable hardware environment to test it.

Keep a copy of your setup if you over-write your complete library.

commit 4cc57699 (2021-03-27): Fix #1100

Fix STM32 + SSD1963 combination
closed 2021-03-27 by Bodmer