Bodmer/TFT_eSPI #1101: Cannot get display to show any example code

Skytek65, 2021-03-25:
I got my board from here:https://www.aliexpress.com/item/32992515344.html?spm=a2g0s.9042311.0.0.56594c4dbTEGkd
And here is the data sheet showing the pin out, and below is my user setup.h file. If anyone can assist with me getting this display up and running that'd be awesome, seems like such a cool little board.

[image]

#define ILI9341_DRIVER

#define TFT_MISO 2
#define TFT_MOSI 15
#define TFT_SCLK 18

#define TFT_CS 27
#define TFT_DC 32
#define TFT_RST 05

#define LOAD_GLCD
#define LOAD_FONT2
#define LOAD_FONT4
#define LOAD_FONT6
#define LOAD_FONT7
#define LOAD_FONT8
#define LOAD_GFXFF

#define SMOOTH_FONT

#define SPI_FREQUENCY  40000000 // Maximum for ILI9341

dronecz, 2021-03-25:
HI @Skytek65, 

your LCD is on second SPI bus so you have to change your settings accordingly: 

#define TFT_MISO 2 -> 23
#define TFT_MOSI 15 -> 12

Let us know if it works ;)

Skytek65, 2021-03-25:
Tried that in the User-Setup.h file sadly still no dice. Here is the example program I'm trying to run, and here is another output from the read user setup test program.

##TFT_Pie_Chart##
`#include <TFT_eSPI.h> // Hardware-specific library
#include <SPI.h>

TFT_eSPI tft = TFT_eSPI(); // Invoke custom library

#define DEG2RAD 0.0174532925

byte inc = 0;
unsigned int col = 0;

void setup(void)
{
  tft.begin();

  tft.setRotation(1);

  tft.fillScreen(TFT_BLACK);
}

void loop() {

  // Draw 4 pie chart segments
  fillSegment(160, 120, 0, 60, 100, TFT_RED);
  fillSegment(160, 120, 60, 30, 100, TFT_GREEN);
  fillSegment(160, 120, 60 + 30, 120, 100, TFT_BLUE);
  fillSegment(160, 120, 60 + 30 + 120, 150, 100, TFT_YELLOW);

  delay(4000);

  // Erase old chart with 360 degree black plot
  fillSegment(160, 120, 0, 360, 100, TFT_BLACK);
}

// #########################################################################
// Draw circle segments
// #########################################################################

// x,y == coords of centre of circle
// start_angle = 0 - 359
// sub_angle   = 0 - 360 = subtended angle
// r = radius
// colour = 16 bit colour value

int fillSegment(int x, int y, int start_angle, int sub_angle, int r, unsigned int colour)
{
  // Calculate first pair of coordinates for segment start
  float sx = cos((start_angle - 90) * DEG2RAD);
  float sy = sin((start_angle - 90) * DEG2RAD);
  uint16_t x1 = sx * r + x;
  uint16_t y1 = sy * r + y;

  // Draw colour blocks every inc degrees
  for (int i = start_angle; i < start_angle + sub_angle; i++) {

    // Calculate pair of coordinates for segment end
    int x2 = cos((i + 1 - 90) * DEG2RAD) * r + x;
    int y2 = sin((i + 1 - 90) * DEG2RAD) * r + y;

    tft.fillTriangle(x1, y1, x2, y2, x, y, colour);

    // Copy segment end to sgement start for next segment
    x1 = x2;
    y1 = y2;
  }
}

// #########################################################################
// Return the 16 bit colour with brightness 0-100%
// #########################################################################
unsigned int brightness(unsigned int colour, int brightness)
{
  byte red   = colour >> 11;
  byte green = (colour & 0x7E0) >> 5;
  byte blue  = colour & 0x1F;

  blue =  (blue * brightness)/100;
  green = (green * brightness)/100;
  red =   (red * brightness)/100;

  return (red << 11) + (green << 5) + blue;
}
`
###############

##Read User Setup##
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4

[code]
TFT_eSPI ver = 2.3.61
Processor    = ESP32
Frequency    = 240MHz
Transactions = Yes
Interface    = SPI
Display driver = 9341
Display width  = 240
Display height = 320

MOSI    = GPIO 12
MISO    = GPIO 23
SCK     = GPIO 18
TFT_CS   = GPIO 27
TFT_DC   = GPIO 32
TFT_RST  = GPIO 5

Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled

Display SPI frequency = 40.00
[/code]
##################

dronecz, 2021-03-25:
Did you try to swap MISO & MOSI pins?

Bodmer, 2021-03-25:
The MOSI line should connect to the display SDI and MISO to SDO.
Correct setup for these pins is therefore:

#define TFT_MISO 12
#define TFT_MOSI 23

Bodmer, 2021-03-25:
There is a ready made setup file in the Users_Setups.h folder called Setup22_TTGO_T4_v1.3.h, t
use that as it includes the back light pin as well.

Bodmer, 2021-03-25:
FYI [here is the setup](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setups/Setup22_TTGO_T4_v1.3.h)

Skytek65, 2021-03-25:
You are the man, appreciate the help and am excited to start working with this!

closed 2021-03-25 by Skytek65