Bodmer/TFT_eSPI #1105: Part of my font is not displayed on ILI9341

ArtGudvin, 2021-03-28:
I created my own font in which I added Russian letters. Russian letters added to the range 0xC0 - 0xFF [FreeSansEnRu9pt8b.zip](https://github.com/Bodmer/TFT_eSPI/files/6217241/FreeSansEnRu9pt8b.zip)

Not all letters are displayed in the TFT_eSPI library. 
(АБВГДЕЖЗИЙКЛМНОПРСТУФХЦЧЪШЩЫЬЭЮЯ) (абвгдежзийклмноп) - Not displayed (UTF8: D090 - D0BF).
(рстуфхцчъшщыьэюя) - Displayed well (UTF8: D180 - D18F)
[image]

```
#include <TFT_eSPI.h> // Graphics and font library for ILI9341 driver chip

TFT_eSPI tft = TFT_eSPI();  // Invoke library, pins defined in User_Setup.h

void setup() {
  // put your setup code here, to run once:
  tft.init();
  tft.setRotation(1);
  tft.fillScreen(TFT_BLACK); //Закрашиваем экран черным
}

void loop() {
  // put your main code here, to run repeatedly:    
    tft.setCursor(0,50);
    tft.setFreeFont(&FreeSansEnRu9pt8b); 
       
    tft.println("ABCDEFGHIJKMNOPRSYXW");
    tft.println("abcdefghijklmnoprstyxw");
    tft.println(utf8rus("АБВГДЕЖЗИЙКЛМНОПРСТУФХЦЧЪШЩЫЬЭЮЯ"));
    tft.println(utf8rus("абвгдежзийклмнопрстуфхцчъшщыьэюя"));
          
    delay(10000);
    tft.fillScreen(TFT_BLACK); 
}

/* Recode russian fonts from UTF-8 to Windows-1251 */

String utf8rus(String source)
{
  int i,k;
  String target;
  unsigned char n;
  char m[2] = { '0', '\0' };

 
  k = source.length(); i = 0;
 
  while (i < k) {
    n = source[i]; i++;
 
    if (n >= 0xC0) {
      switch (n) {
        case 0xD0: {    
            n = source[i]; i++;
            if (n == 0x81) {
              n = 0xA8;
              break;
            }
            if (n >= 0x90 && n <= 0xBF) n = n + 0x30;      
            break;
          }
        case 0xD1: { 
            n = source[i]; i++;
            if (n == 0x91) {
              n = 0xB8;
              break;
            }      
            if (n >= 0x80 && n <= 0x8F) n = n + 0x70; 
            break;
          }
      }
    } 
    m[0] = n; target = target + String(m);
  }
return target;
 
}

```

All letters are displayed correctly in the Adafruit library.

[image]

```
#include "Adafruit_GFX.h"    //Дисплей ILI 9431 320x240
#include "Adafruit_ILI9341.h"  //Дисплей ILI 9431 320x240
#include <Fonts/FreeSansEnRu9pt8b.h>    

#define TFT_BLACK ILI9341_BLACK//BLACK
#define TFT_DC 0    //D3
#define TFT_CS 15   //D8

Adafruit_ILI9341 tft = Adafruit_ILI9341(TFT_CS, TFT_DC);

void setup() {
  // put your setup code here, to run once:
  tft.begin();
  tft.cp437(true);
  tft.setRotation(1);
  tft.fillScreen(TFT_BLACK); //Закрашиваем экран черным
}

void loop() {
  // put your main code here, to run repeatedly:

    tft.setCursor(0,50);
    tft.setFont(&FreeSansEnRu9pt8b); 
    
    tft.println("ABCDEFGHIJKMNOPRSYXW");
    tft.println("abcdefghijklmnoprstyxw");
    tft.println(utf8rus("АБВГДЕЖЗИЙКЛМНОПРСТУФХЦЧЪШЩЫЬЭЮЯ"));
    tft.println(utf8rus("абвгдежзийклмнопрстуфхцчъшщыьэюя"));
          
    delay(10000);
    tft.fillScreen(TFT_BLACK); //Закрашиваем экран черным

}

/* Recode russian fonts from UTF-8 to Windows-1251 */

String utf8rus(String source)
{
  int i,k;
  String target;
  unsigned char n;
  char m[2] = { '0', '\0' };

 
  k = source.length(); i = 0;
 
  while (i < k) {
    n = source[i]; i++;
 
    if (n >= 0xC0) {
      switch (n) {
        case 0xD0: {
            n = source[i]; i++;
            if (n == 0x81) {
              n = 0xA8;
              break;
            }
            if (n >= 0x90 && n <= 0xBF) n = n + 0x30;
            break;
          }
        case 0xD1: {
            n = source[i]; i++;
            if (n == 0x91) {
              n = 0xB8;
              break;
            }
            if (n >= 0x80 && n <= 0x8F) n = n + 0x70;
            break;
          }
      }
    } 
    m[0] = n; target = target + String(m);
  }
return target;
 
}
```

Tell me what could be the problem?

Bodmer, 2021-03-28:
The Adafruit library does not handle UTF8 encoded strings and so it looks like you have created a work-around.

The TFT_eSPI library does handle UTF8 and decoding is enabled by default.

You can switch of UTF8 decoding by using:
```
  tft.setAttribute(UTF8_SWITCH, 0);    // Turn off
 // tft.setAttribute(UTF8_SWITCH, 1);  // Turn on 
```

Try that, or use the UTF8 decoding already built in

Bodmer, 2021-03-28:
I checked the code and the UTF8 switch has not been fully implemented.

To switch OFF UTF8 decoding you will need to [comment out this line](https://github.com/Bodmer/TFT_eSPI/blob/master/TFT_eSPI.cpp#L3409), thus:
```
//#define DECODE_UTF8 // Test only, comment out to stop decoding
```
Your TFT_eSPI sketch above should then work OK without any changes as you have moved the characters in the character map.

You may also be interested in using smooth fonts, they allow any mix of 16 bit Unicode glyphs.

ArtGudvin, 2021-03-29:
Thanks, Bodmer.

Indeed, there is no difference when switching:
```
tft.setAttribute(UTF8_SWITCH, 0);    // Turn off
 // tft.setAttribute(UTF8_SWITCH, 1);  // Turn on 
```
But it worked:
`//#define DECODE_UTF8 // Test only, comment out to stop decoding`

Bodmer, 2021-03-29:
Great! Hope your project is a success.

closed 2021-03-29 by Bodmer

lepromer, 2022-07-19:
THANK YOU @Bodmer in my project this code was worked