Bodmer/TFT_eSPI #1106: ST7796S / yellow tint issue

Koxx3, 2021-03-28:
Hello,

as I finally found a 3.5" with SPI and ST7796S (as you advised me).
The controller really rocks ! 
40MHz / 80MHz pass with the ESP32 !

This is this screen :
https://fr.aliexpress.com/item/1005001797227781.html?spm=a2g0s.9042311.0.0.7fce6c37kSkbrg

BUT I have a major issue with colors... 
It's tinted in YELLOW (on both screens I purchased).

Could it be a software issue ?

Cuold I solve this by software (color tint filter) ?

Koxx3, 2021-03-28:
it's an IPS screen.

Koxx3, 2021-03-28:
[image]

Koxx3, 2021-03-28:
[image]

Koxx3, 2021-03-28:
it's difficult to see. but even from the top, really a yellow tint :(

Koxx3, 2021-03-28:
oh yes ... I use SPI interface... if it matters

Koxx3, 2021-03-28:
I see some gamma in the init sequence. could it be related ? how to change that ?

Koxx3, 2021-03-28:
the vendor have sent me this code for SPI 4 wires.

I've replaed the gamma part... I don't seen any improvement.
I am checking the rest.

```

//NOTE£ºVCC=3.3V, IOVCC=1.8-3.3V
Display resolution:320*480
params->dsi.vertical_sync_active=2
params->dsi.vertical_backporch=8
params->dsi.vertical_frontporch=4
params->dsi.horizontal_sync_active=2
params->dsi.horizontal_backporch=8
params->dsi.horizontal_frontporch=4
params->RGB_CLOCK=(10)M
Frame Rate=60HZ
//****************************************************************
void write_command (unsigned char y) // (uchar y,uchar x)
{
  unsigned char i;
  cs=0;
  rs=0;
  for(i=0;i<8;i++)
    {
     scl=0;
      if (y&0x80)
          sdi=1;
          else
          sdi=0;
       scl=1;
      y=y<<1;
    }
cs=1;
}
//************************************Êý¾Ý***************************
void write_data(unsigned char w) // (uchar w, uchar v)
{
  unsigned char i;
cs=0;
rs=1;
for(i=0;i<8;i++)
 {
  scl=0;
     if (w&0x80)
        sdi=1;
         else
         sdi=0;
        scl=1;
      w=w<<1;
    }
cs=1;
}
/*************************************************/
void initi(void)
{
   res=1;
   delay(1);
   res=0;
   delay(10);
   res=1;
   delay(120);
/***************************************/ 
write_command(0xF0); //Command Set Control       
write_data(0xC3);   

write_command(0xF0); //Command Set Control     
write_data(0x96);   

write_command(0x36);//Memory Data Access Control        
write_data(0x48);   

write_command(0x3A);//Interface Pixel Format        
write_data(0x50);//65K  

write_command(0xB4); //1-dot Inversion
write_data(0x01);  //01  1dot   00 column  02 2ot  

write_command(0xB1);//FRMCTR1     
write_data(0x80);   //FRS[D7-D4], DIVA[D1-D0] 81 for 15Hz
write_data(0x10);   //RTNA[4:0]

 write_command(0xB6); 
write_data(0x20); //set rgb interface and DE mode
write_data(0x02); //GS,SS
write_data(0x3B);

write_command(0XB0);  //Interface Mode Control  
write_data(0x00); 

write_command(0xC0);//Power Control 1      
write_data(0x80);  //AVDD ,AVCL  									   `
write_data(0x64); //VGH=15V VGL=-10V  

write_command(0xC1);  //Power Control 2        
write_data(0x13);  //VOP=4.5V

write_command(0xC2); //Power Control 3      
write_data(0xA7);   

write_command(0xC5);//VCOM Control      
write_data(0x10);   

write_command(0xE8); //Display Output Ctrl Adjust     
write_data(0x40);   
write_data(0x8a);   
write_data(0x00);   
write_data(0x00);   
write_data(0x29);   
write_data(0x19);   
write_data(0xA5);   
write_data(0x33);   

write_command(0xE0);//GAMMA
write_data(0xF0);
write_data(0x06);
write_data(0x0B);
write_data(0x07);
write_data(0x06);
write_data(0x05);
write_data(0x2E);
write_data(0x33);
write_data(0x47);
write_data(0x3A);
write_data(0x17);
write_data(0x16);
write_data(0x2E);
write_data(0x31);

write_command(0xE1);//GAMMA
write_data(0xF0);
write_data(0x09);
write_data(0x0D);
write_data(0x09);
write_data(0x08);
write_data(0x23);
write_data(0x2E);
write_data(0x33);
write_data(0x46);
write_data(0x38);
write_data(0x13);
write_data(0x13);
write_data(0x2C);
write_data(0x32);

write_command(0xF0); //Command Set Control     
write_data(0x3C);   

write_command(0xF0);//Command Set Control      
write_data(0x69);  

write_command(0x21); //Enter_invert_mode 
 //**********************320*480**********************  
write_command(0x2A);    //320 
write_data(0x00);   
write_data(0x00);   
write_data(0x01);   
write_data(0x3F);   

write_command(0x2B);    //480
write_data(0x00);   
write_data(0x00);   
write_data(0x01);   
write_data(0xDF);  

write_command(0x11); // Exit Sleep    
 delay(120);  
               //ms
write_command(0x29);//DISPLAY ON       
 delay(50); 
write_command(0x2c); //Write memory start   
	   
}
//******************************************
void LCD_Enter_Standby(void)
{
   write_command(0x28);//display off
   delay(10);
  write_command(0x10);
   delay(120);
   RGB_OFF();
   }
void LCD_Exit_Standby (void)
{
   RGB_ON();
   delay(10);
   write_command(0x11);// Exit Sleep/ Standby mode
   delay(120);
   write_command(0x29);
  }

```

Bodmer, 2021-03-28:
Normally these displays are ST7796S but I see the one you have is ST7796I - I am not sure if that means the TFT array is different.

It looks like the blue is not bright enough, the red and green then combine to produce a yellow hue.

In addition to the Gamma you might have to change other registers such as the internal voltages, try using the values provided by the vendor:

```
writecommand(0xC0);//Power Control 1      
writedata(0x80);  //AVDD, AVCL
writedata(0x64); //VGH=15V VGL=-10V  

writecommand(0xC1);  //Power Control 2        
writedata(0x13);  //VOP=4.5V

writecommand(0xC2); //Power Control 3      
writedata(0xA7);   

writecommand(0xC5);//VCOM Control      
writedata(0x10); 
```

Koxx3, 2021-03-28:
thanks a lot for your answer.
Yes, it's a little better. still yellow-ish... but better.  ;)

I took most of his code.

Is it possible to keep the lib intact and use a custom init ?

Koxx3, 2021-03-28:
[image]

Koxx3, 2021-03-28:
I cheat, I don't use a "pure white" ;)

```
#define WHITE_RED 0xe0
#define WHITE_GREEN 0xe0
#define WHITE_BLUE 0xff
#define MY_TFT_WHITE (((WHITE_RED & 0xf8)<<8) + ((WHITE_GREEN & 0xfc)<<3)+(WHITE_BLUE>>3))
```

Bodmer, 2021-03-28:
That is a good solution. There is a function to convert RGB colours to 16 bit that you can use:
```
           // Convert 8 bit red, green and blue to 16 bits
  uint16_t color565(uint8_t red, uint8_t green, uint8_t blue);
```

Bodmer, 2021-03-28:
Post the final init code you use and I can add it as an option.

Koxx3, 2021-03-29:
it's here ;)

https://github.com/Koxx3/SmartController_SmartDisplay_ESP32/blob/master/documents/screen/3.5v2%20st7796/tft_espi_init_new.h

Bodmer, 2021-03-29:
Thanks. I will add in the next update cycle.

closed 2021-03-29 by Bodmer