Bodmer/TFT_eSPI #1113: ST7735_BLACKTAB flicker ~ another variation of display hardware?

giantclambake, 2021-04-01:
I have been working on a project that uses a TFT display of this type; 

https://www.hotmcu.com/18-128x160-tft-lcd-with-spi-interface-p-314.html

The display I have been using has a green tab, however it only works correctly with ST7735_BLACKTAB defined.

I ordered some more of this display type, and when they arrived it turns out they are black tab types. However, they do not work correctly with ST7735_BLACKTAB being defined (nor any of the other valid options)

This issue is a white flicker, extending from the left side of display, that varies in intensity/width as the sketch is running.

Issue is able to be recreated by running the example 'TFT_Clock_Digital' included with TFT_eSPI on one of these black tab displays ( I have 3 of them here and they all behave the same)

Note: if SPI_FREQUENCY is increased to 40000000 this white flickering stops altogether -- of course however, fonts are trashed and screen artifacts aplenty at this value. (normal SPI_FREQUENCY setting is 27000000)...but the flicker disappears.

The same type display I have here which is a greentab variety that only works with ST7735_BLACKTAB defined, displays no such issues as detailed above.

Visually, you can watch this vid of the misbehavior ->   https://photos.app.goo.gl/3pWaoxgQPFgot9vC8

I can only surmise these black tab displays I've purchased, represent yet another variant of the ST7735 implementation?

Any advice on how to proceed towards fixing this issue appreciated ; if you require an example of the TFT display type in question here, I am prepared to send same to you gratis  =)

Kudos for the great library,

Don

giantclambake, 2021-04-02:
Follow-up:

I should have looked at SPI_FREQUENCY more closely ; after further testing, I have concluded the following;

*Original GREEN tab display only works correctly with ST7735_BLACKTAB defined has a max SPI_FREQUENCY=27000000

*Recently bought BLACK tab display(s) with ST7735_BLACKTAB defined has a max SPI_FREQUENCY=10000000

In User_Setup.h appears the following text;

// Define the SPI clock frequency, this affects the graphics rendering speed. Too
// fast and the TFT driver will not keep up and display corruption appears.
// With an ILI9341 display 40MHz works OK, 80MHz sometimes fails
// With a ST7735 display more than 27MHz may not work (spurious pixels and lines)
// With an ILI9163 display 27 MHz works OK.

I feel the words 'display corruption' slightly misled me ...ie; no characters were actually being corrupted/misdrawn, as can be seen in the above video link, the problem presents itself more as 'display flickering' without any apparent artifacts/corruption. After testing 3 of these displays (same results for each), I feel it would be safe to add to the above text, a line reading;

//With some ST7735 'BLACKTAB' displays more than 10MHz may not work correctly and cause display flickering

In any event, feel free to close this issue -- the above line summarizes the problem, and the ST7735 driver works correctly on these black tab displays with the reduced SPI_FREQUENCY value. If you feel the driver code needs revising to address this (it very well may just be the specific display hardware), previous offer still stands.

Thanks, and keep up the good work!

Don

Bodmer, 2021-04-03:
Display corruption is usually a gradual process as SPI frequency increases. As the SPI frequency increases further the display does not get initialised so then ends up staying blank.

I suspect the second display batch have a clone copy chip  but this is a guess as others have noted that some chips do not have expected chip ID. However since 37MHz is over locking the genuine chips anyway we can't complain.

closed 2021-04-03 by Bodmer