Bodmer/TFT_eSPI #1124: pushImage transparent not working. Need transparent instead TFT_WHITE

dsiberia9s, 2021-04-10:
// const unsigned char player_1[1320] = {0xff, 0xff, 0xff, ....

void setup() {
  M5Stack();
  tft.pushImage(0, 0, 320, 240, (const uint16_t*)img_1);
  s.createSprite(40, 66);
}

void loop() {
  for (int i = 0; i < 160; i++) {    
    s.pushImage(10, 15, 20, 33, (uint16_t*)player_1, TFT_WHITE);
    s.pushSprite(i, 10);
    delay(40);
  }
}

[image]

Bodmer, 2021-04-10:
The transparent colour needs to be specified when pushing to the screen, try:
```
s.pushSprite(i, 10, TFT_WHITE);
```
Altough you can push an image to a sprite with a transparent colour I suspect tis line could be used instead with a slight performance advantage:
```
s.pushImage(10, 15, 20, 33, (uint16_t*)player_1);
```

dsiberia9s, 2021-04-10:
```
void loop() {
  for (int i = 0; i < 160; i++) {    
    s.pushImage(10, 15, 20, 33, (uint16_t*)player_1);
    s.pushSprite(i, 10, TFT_WHITE);
    delay(40);
  }
}
```

[image]

Bodmer, 2021-04-10:
Looks like it is working now. Since parts on the image are not white they will not draw as transparent.

closed 2021-04-10 by Bodmer

dsiberia9s, 2021-04-10:
Understand. Thanks!