Bodmer/TFT_eSPI #1127: How to remove traces of an image?

dsiberia9s, 2021-04-10:
```
void loop() {
  for (int i = 0; i < 160; i++) {    
    s.pushImage(30, 30, 60, 60, (uint16_t*)player_1);
    s.pushSprite(i, 10, TFT_BLACK);
    delay(40);
  }
}
```

[image]

Bodmer, 2021-04-10:
This is expected, see discussion [starting here](url). And the [explanation here](https://github.com/Bodmer/PNG_TEST_ONLY/issues/1#issuecomment-816009274).

dsiberia9s, 2021-04-11:
String s.pushImage(10, 10, 64, 64, (uint16_t*)player_1, TFT_BLACK); not provide transparent for image.

```
void loop() {
  int x = 50;

  for (; x < 180; x+=2) {
    s.pushImage(-x, 0, 320, 240, (uint16_t*)img_1);
    s.setSwapBytes(true);
    s.pushImage(10, 10, 64, 64, (uint16_t*)player_1, TFT_BLACK);
    s.setSwapBytes(false);
    //s.drawRect(10, 10, 60, 60, TFT_RED);
    s.pushSprite(x, 0);
    delay(40);
  }
  s.pushImage(-x, 0, 320, 240, (uint16_t*)img_1);
  s.pushSprite(x, 0, TFT_BLACK);
  
}
```

Bodmer, 2021-04-11:
There is no pushImage() with a transparent colour in the Sprite class. So due to class inheritance the TFT function is substitued which pushes direct to the TFT. I know this is confusing but I do not have the time to implement all possible permutations and combinations that folk may want! To handle sprites and the complexity of different bits per pixel, the image will first have to be pushed into a dedicated sprite, then this sprite pushed into another one with transparency [using this function](https://github.com/Bodmer/TFT_eSPI/blob/master/Extensions/Sprite.cpp#L741-L752), the library then handles the bpp translation for the cases listed.

dsiberia9s, 2021-04-11:
[image]

OMG! It's work! Thank you so much! 👍🏻👍🏻

`void loop() {
  int x = 50;

  for (; x < 180; x+=2) {
    s.pushImage(-x, 0, 320, 240, (uint16_t*)img_1);

    transp_sprite.setSwapBytes(true);
    transp_sprite.pushImage(10, 10, 64, 64, (uint16_t*)player_1);
    transp_sprite.setSwapBytes(false);
    transp_sprite.pushToSprite(&s, 0, 0, TFT_BLACK);
    
    //s.drawRect(10, 10, 60, 60, TFT_RED);
    s.pushSprite(x, 0);
    delay(40);
  }
  s.pushImage(-x, 0, 320, 240, (uint16_t*)img_1);
  s.pushSprite(x, 0, TFT_BLACK);
  
}`

closed 2021-04-11 by dsiberia9s

Bodmer, 2021-04-11:
Glad to here it is working.