Bodmer/TFT_eSPI #1131: ESP 32 GPIO availability

3mc2, 2021-04-13:
Hello,

I was wondering why I am able to use only GPIO 0 - 31 on the ESP32.  There are 2 more available GPIO (32,33) which are available.  In the application I am doing I am out of GPIO except 0, and using that creates all kinds of problems.  If you could tell me why GPIO 32,33 are not available?

Thanks.
PS.  This is really a great library.

Bodmer, 2021-04-13:
There are only restrictions for certain functions (DC, WR, D0-7 ) with the parallel interface so that a single register write can be used, this boosts performance. So only 10 pins in this particular case has a restriction on allocation.

The other pins can be allocated to other functions, [see here](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setup.h#L238-L253), where 32 and 33 are used.

closed 2021-04-13 by Bodmer

3mc2, 2021-04-13:
Ah.  I thought the limitation was across all pins.  What you are saying
makes sense.

Thank you for the clarification.